openwrt/packages #26576: perl-authen-sasl-xs: assignment to 'int (*)(void)' from incompatible pointer type

BKPepe, 2025-05-23:
# Package info
Maintainer: @pprindeville 
OpenWrt snapshots, arm_cortex-a9+vfpv3-d16_musl

# Log:
```
XS.xs: In function 'AddCallback':
XS.xs:884:39: warning: format '%x' expects argument of type 'unsigned int', but argument 2 has type 'long unsigned int' [-Wformat=]
  884 |                                 croak("Unknown parameter to %x callback.\n", cb->id);
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~
      |                                                                                |
      |                                                                                long unsigned int
XS.xs:895:43: error: assignment to 'int (*)(void)' from incompatible pointer type 'int (*)(void *, int,  const char **, unsigned int *)' [-Wincompatible-pointer-types]
  895 |                                  cb->proc = PerlCallback;
      |                                           ^
XS.xs:899:42: error: assignment to 'int (*)(void)' from incompatible pointer type 'int (*)(sasl_conn_t *, void *, int,  sasl_secret_t **)' {aka 'int (*)(struct sasl_conn *, void *, int,  struct sasl_secret **)'} [-Wincompatible-pointer-types]
  899 |                                 cb->proc = PerlCallbackSecret;
      |                                          ^
XS.xs:903:42: error: assignment to 'int (*)(void)' from incompatible pointer type 'int (*)(void *, int,  const char **, const char **)' [-Wincompatible-pointer-types]
  903 |                                 cb->proc = PerlCallbackRealm;
      |                                          ^
XS.xs:910:42: error: assignment to 'int (*)(void)' from incompatible pointer type 'int (*)(sasl_conn_t *, void *, const char *, unsigned int,  const char *, unsigned int,  const char *, unsigned int,  struct propctx *)' {aka 'int (*)(struct sasl_conn *, void *, const char *, unsigned int,  const char *, unsigned int,  const char *, unsigned int,  struct propctx *)'} [-Wincompatible-pointer-types]
  910 |                                 cb->proc = PerlCallbackAuthorize;
      |                                          ^
XS.xs:914:42: error: assignment to 'int (*)(void)' from incompatible pointer type 'int (*)(sasl_conn_t *, void *, const char *, unsigned int,  unsigned int,  const char *, char *, unsigned int,  unsigned int *)' {aka 'int (*)(struct sasl_conn *, void *, const char *, unsigned int,  unsigned int,  const char *, char *, unsigned int,  unsigned int *)'} [-Wincompatible-pointer-types]
  914 |                                 cb->proc = PerlCallbackCanonUser;
      |                                          ^
XS.xs:918:42: error: assignment to 'int (*)(void)' from incompatible pointer type 'int (*)(sasl_conn_t *, void *, const char *, const char *, unsigned int,  struct propctx *)' {aka 'int (*)(struct sasl_conn *, void *, const char *, const char *, unsigned int,  struct propctx *)'} [-Wincompatible-pointer-types]
  918 |                                 cb->proc = PerlCallbackServerCheckPass;
      |                                          ^
XS.xs:922:42: error: assignment to 'int (*)(void)' from incompatible pointer type 'int (*)(sasl_conn_t *, void *, const char *, const char *, unsigned int,  struct propctx *, unsigned int)' {aka 'int (*)(struct sasl_conn *, void *, const char *, const char *, unsigned int,  struct propctx *, unsigned int)'} [-Wincompatible-pointer-types]
  922 |                                 cb->proc = PerlCallbackServerSetPass;
      |                                          ^
In file included from /builder/shared-workdir/build/sdk/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/perl5/5.40/CORE/perl.h:6166,
                 from XS.xs:43:
XS.xs: In function 'XS_Authen__SASL__XS_property':
XS.xs:1886:40: error: initialization of 'IV' {aka 'long int'} from 'int *' makes integer from pointer without a cast [-Wint-conversion]
 1886 |                                 XPUSHi((int *)value);
      |                                        ^
/builder/shared-workdir/build/sdk/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/perl5/5.40/CORE/pp.h:484:23: note: in definition of macro 'TARGi'
  484 |         IV TARGi_iv = i;                                                \
      |                       ^
XS.xs:1886:33: note: in expansion of macro 'XPUSHi'
 1886 |                                 XPUSHi((int *)value);
      |                                 ^~~~~~
make[4]: *** [Makefile:343: XS.o] Error 1
```

Full log: https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a9_vfpv3-d16/packages/perl-authen-sasl-xs/compile.txt

pprindeville, 2025-05-24:
@BKPepe Is this another case of it just failing on that one architecture?  What's different about it?  Is it a 32-bit architecture?  Older version of gcc?

BKPepe, 2025-05-24:
No, I've checked multiple targets and it is failing there with the same issue as well.

- [aarch64_cortex-a53](https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/perl-authen-sasl-xs/compile.txt)

- [mipsel_24kc](https://downloads.openwrt.org/snapshots/faillogs/mipsel_24kc/packages/perl-authen-sasl-xs/compile.txt)

- [x86_64](https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/perl-authen-sasl-xs/compile.txt)

- [powerpc_8545](https://downloads.openwrt.org/snapshots/faillogs/powerpc_8548/packages/perl-authen-sasl-xs/compile.txt)

Most likely, the perl-auten-sasl-xs does not support GCC 14.

pprindeville, 2025-05-24:
> [@pprindeville](https://github.com/pprindeville) Most likely, you will need to backport this patch: https://sources.debian.org/patches/libauthen-sasl-xs-perl/1.00-3/gcc-14.patch/
> 
> and while looking at Debian's page, there are also a few more patches, which you might want to backport.

I don't have time to do a triage (I've got a Starlink Mini I need to put on the roof plus a GPS antenna for a Meinberg PCI-e card) but if you have a list I'll see what I can do.

jw2013, 2025-06-10:
@pprindeville , may I help you maintaining this package?

jw2013, 2025-06-10:
Ok, I'll check and test the various Debian patches above, and prepare the PR accordingly.

pprindeville, 2025-06-10:
> [@pprindeville](https://github.com/pprindeville) , may I help you maintaining this package?

That would be great.  Perl takes as much time as all of the other packages I maintain combined...

jw2013, 2025-06-10:
Hmm... I somewhat doubt that the former package worked at all before:
- perlbase-dynaloader was missing in DEPENDS (but might have been available because of other packages), and
- the resulting package file did NOT include the compiled library auto/Authen/SASL/XS/XS.so (!)

I omitted the patches from Debian, that relate to documentation and testing only. The one with number 106 only includes the first half of the Debian one, which fixes something from 105. Hope it was ok to keep them as 2 separate ones?