openwrt/packages #26582: bottom 0.9.4 requires newer rustc

BKPepe, 2025-05-23:
While checking faillogs, I can see that bottom package is not compiled for OpenWrt 23.05.
```
Log:
error: failed to compile `bottom v0.9.4 (/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/bottom-0.9.4)`, intermediate artifacts can be found at `/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/bottom-0.9.4/target`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.

Caused by:
  rustc 1.81.0 is not supported by the following package:
    backtrace@0.3.75 requires rustc 1.82.0
  Either upgrade rustc or select compatible dependency versions with
  `cargo update <name>@<current-ver> --precise <compatible-ver>`
  where `<compatible-ver>` is the latest version supporting rustc 1.81.0
make[3]: *** [Makefile:42: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/bottom-0.9.4/.built] Error 101
time: package/feeds/packages/bottom/compile#0.81#0.50#1.93
```

Full log is available here: https://downloads.openwrt.org/releases/faillogs-23.05/aarch64_cortex-a53/packages/bottom/compile.txt

cc: maintainer: @lu-zero

lu-zero, 2025-05-23:
I guess it is a good time to bump

lu-zero, 2025-05-23:
but currently we should have 1.85.0 it wasn't brought on the 23.05 branch?

1715173329, 2025-05-28:
> As you prefer :) I'd rather use always the latest stable at least for now if we do not have reasons not to.

Well, I personally prefer to lock dependencies for this package as 23.05 was released 2 years ago and might EoL soon. This can avoid some potential regressions.

PR #26614 (2025-05-31): [openwrt-23.05] rust: update to 1.85.0
Maintainer: @lu-zero
Compile tested: rockchip/armv8, tested with bottom and netavark
Run tested: nanopi-r5c (bottom)

Description:
Update rust to 1.85.0

Fixes: #26582