openwrt/packages #26618: `docker stats` cannot obtain cpu usage and mem usage.

chunfengyao, 2025-05-30:
When executing `docker stats`, CPU usage and memory usage always show `0%, 0B/0B`. Additionally, Net I/O metrics show `0B/0B` if the container uses the host network. The Home Assistant Docker Monitor plugin also couldn’t retrieve this information from the mapped `/var/run/docker.sock`. This issue appears to start in version `24.10.1`.
Is this a bug? Has it been fixed?

[image]

chunfengyao, 2025-05-30:
[image]
[image]
Screenshot from HomeAssistant.

brada4, 2025-05-30:
Could you run diffconfig script and reduce config output?

chunfengyao, 2025-10-31:
@brada4 Could you please ​look at​ this issue and the log file I uploaded ​again​? or could you tag the appropriate person to take a look?

brada4, 2025-10-31:
You can remount cgroupfs between cgroup and cgroup2 type. Docker is still against the later while everything else supports that v2

chunfengyao, 2025-11-01:
The issue did not reappear after I removed `cgroup-mount` and `cgroup-tools`. which suggests that `cgroup-mount` had created a mixed runtime environment incompatible with Docker. Thank you. @brada4

closed 2025-11-01 by chunfengyao