openwrt/packages #26623: rust 1.87.0 compile error llvm.download-ci-llvm cannot be set to true

smallprogram, 2025-05-30:
when the compilation is executed locally, there is no problem, no error, and it can be compiled perfectly. However, when using the Ubuntu-latest system environment (Ubuntu-24.04) in Github Action, an error will appear：
```
thread 'main' panicked at src/bootstrap/src/core/config/config.rs:3166:21:
llvm.download-ci-llvm cannot be set to true on CI. Use if-unchanged instead.
note: run with RUST_BACKTRACE=1 environment variable to display a backtrace
Build completed unsuccessfully in 0:00:00
make[3]: *** [Makefile:108: /home/runner/work/auto-build-openwrt/auto-build-openwrt/openwrt/build_dir/target-x86_64_musl/host/rustc-1.87.0-src/.built] Error 1
make[3]: Leaving directory '/home/runner/work/auto-build-openwrt/auto-build-openwrt/openwrt/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#0.47#0.28#0.74
ERROR: package/feeds/packages/rust [host] failed to build.
```

When I used the prompts and modified it according to the prompts, setting llvm.download-ci-llvm to if-unchanged, it still had another error:
```
ERROR: 'if-unchanged' is only compatible with Git managed sources.
Build completed unsuccessfully in 0:01:04
make[2]: *** [Makefile:108: /workdir/openwrt/build_dir/target-x86_64_musl/host/rustc-1.87.0-src/.built] Error 1
make[2]: Leaving directory '/workdir/openwrt/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#106.84#36.15#109.22
ERROR: package/feeds/packages/rust [host] failed to build.
```

lu-zero, 2025-05-31:
It was mentioned also in https://github.com/openwrt/packages/commit/95eef0fd580a411bd487ccf61a8d4bd25beab5c2#commitcomment-158611911 

Either unset GITHUB_ACTIONS or we should discuss with upstream to disable that test in a cleaner way.

workingjubilee, 2025-05-31:
Please unset `download-ci-llvm`. It is a convenience aimed at developers working on rustc, it is not really intended for other audiences and it's definitely not intended for other people's CI. That's why it broke your job.