openwrt/packages #26690: MWAN3 with IPv6 don't come back IPv6 interface after address change

victor186, 2025-06-04:
### Package Name

Mwan3

### Maintainer

idk

### OpenWrt Version

24.10.1

### OpenWrt Target/Subtarget

ramips/mt7621

### Steps to Reproduce

Connect the openwrt router tô a LTE modem
Reset the LTE interface on modem to change IPv6 address
The mwan3 will go to offline state on the interface and don't come back before you restart interface manually, but i can ping IPv6 tracking address from router.

### Actual Behaviour

[image]
[image]

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

victor186, 2025-06-04:
My IPV6 interfaces on mwan config: 

`config interface 'MF286C'
        option enabled '1'
        option initial_state 'online'
        option family 'ipv6'
        option track_method 'ping'
        option reliability '1'
        option count '1'
        option timeout '4'
        option interval '10'
        option failure_interval '5'
        option recovery_interval '5'
        option down '5'
        option up '5'
        list track_ip '64:ff9b::8efb:818e'
        list track_ip '64:ff9b::0101:0101'
        option size '56'
        option max_ttl '60'

config interface 'V6mdm'
        option enabled '1'
        option initial_state 'online'
        option family 'ipv6'
        option track_method 'ping'
        option reliability '1'
        option count '1'
        option timeout '4'
        option interval '3'
        option failure_interval '3'
        option recovery_interval '1'
        option down '3'
        option up '3'
        list track_ip '2606:4700:4700::1111'
        option size '56'
        option max_ttl '60'`

feckert, 2025-06-05:
Is it really correct that the logical `MF286C` interface uses the physical `wan` device?
This is about the mobile service device. This normally uses the physical interface `wwan`?

oshirinap, 2025-07-03:
Hello, I’m experiencing the same issue:

OpenWrt 23.05.5 (rockchip/armv8)
mwan3 2.11.8-2
IPv6 works normally, but mwan3track fails the ping test. My workaround is to remove all tracking IPs (so tracking is disabled, but this means outages won’t be detected automatically).

Thanks!