openwrt/packages #26696: ath12k-firmware-wcn7850: missing file firmware-2.bin

davygrvy, 2025-06-05:
### Package Name

ath12k-firmware-wcn7850

### Maintainer

@all

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

x86/64

### Steps to Reproduce

1. boot system
2. logread| grep procd
3. Thu Jun  5 13:48:09 2025 daemon.err procd: Could not find firmware /lib/firmware/ath12k/WCN7850/hw2.0/firmware-2.bin: No such file or directory

### Actual Behaviour

file loads and is included is package

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

davygrvy, 2025-06-05:
```
# lspci -nnk | grep Qualcomm
lspci: Unable to load libkmod resources: error -2
04:00.0 Network controller [0280]: Qualcomm Technologies, Inc WCN785x Wi-Fi 7(802.11be) 320MHz 2x2 [FastConnect 7800] [17cb:1107] (rev 01)
```

davygrvy, 2025-06-05:
package version is 20250509-r1

davygrvy, 2025-06-06:
Does the mainline even have this yet?
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/ath12k/WCN7850/hw2.0

BKPepe, 2025-06-06:
> * [x]  I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.

Because this issue does not belong here, I am closing it.
Package Makefile can be found here: https://github.com/openwrt/openwrt/blob/main/package/firmware/linux-firmware/qca_ath12k.mk

closed 2025-06-06 by BKPepe

davygrvy, 2025-06-06:
But there is nothing wrong with the makefile.  The file is missing and therefore can't be copied

davygrvy, 2025-06-06:
banana-pi references it [here](https://forum.banana-pi.org/t/loading-ath12k-kernel-module-qcncm865-card-makes-system-reboot/19270/2) with 

`2024-10-15T20:08:06.389554+01:00 bpi-r4 kernel: [ 1681.724674] ath12k_pci 0001:01:00.0: boot firmware request ath12k/WCN7850/hw2.0/firmware-2.bin size 15224888`