openwrt/packages #26763: mstflint: fails to build with NLS

BKPepe, 2025-06-13:
### Package Name

mstflint

### Maintainer

@tk154

### OpenWrt Version

snapshot and OpenWrt 24.10

### OpenWrt Target/Subtarget

mvebu/cortex-a9 (Turris Omnia)

### Steps to Reproduce

1. Enable CONFIG_BUILD_NLS
2. Try to compile package

### Actual Behaviour

It fails with following output:
```
checking for xmlDocDumpFormatMemoryEnc in -lxml2... no
configure: error: cannot find xml2 xmlDocDumpFormatMemoryEnc() function. Try re-installing the library...
make[2]: *** [Makefile:145: /Volumes/OpenWrt/omnia-hbd/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/mstflint-4.29.0/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
```

I was able to fix this by adding this row to Makefile:
```
include $(INCLUDE_DIR)/nls.mk
```

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

BKPepe, 2025-06-13:
cc: @Zatharalex , who found this and it would be better than refreshing patches every time in your repo. :)
REF: https://gitlab.nic.cz/turris/os/build/-/merge_requests/810

tk154, 2025-06-14:
Thanks for the report. I was able to reproduce the issue, and your provided fix also solved the compilation error for me.

PR #26774 (2025-06-15): mstflint: add missing nls.mk include

## 📦 Package Details

**Maintainer:** me

**Description:**
If `nls.mk` is not included and `BUILD_NLS` is set compilation will fail with various undefined references to the `libiconv` library. So this commit includes the missing `nls.mk`.
This fixes #26763.

---

## 🧪 Run Testing Details

- **OpenWrt Version:** OpenWrt Snapshot
- **OpenWrt Target/Subtarget:** x86/64
- **OpenWrt Device:** Mellanox Spectrum SN2100

---

## ✅ Formalities

- [x] I have reviewed the [CONTRIBUTING.md](../CONTRIBUTING.md) file for detailed contributing guidelines.

### If your PR [... cut]
closed 2025-06-15 by BKPepe