openwrt/packages #26794: syslog-ng: openobserve-log() missing http module

skaa, 2025-06-18:
### Package Name

syslog-ng

### Maintainer

BKPepe

### OpenWrt Version

24.10.1

### OpenWrt Target/Subtarget

rockchip/armv8

### Steps to Reproduce

1. Install syslog-ng package
2. Config /etc/syslog-ng.conf :
```
destination d_openobserve_http {
    openobserve-log(
        url("http://some.lan:5080")
        organization("default")
        stream("syslog-ng")
        user("user@example.com")
        password("password")
    );
};

log {
        source(src);
        source(net);
        source(kernel);
        destination(messages);
        destination(d_openobserve_http);
        flags(flow-control);
};
```
3.  Execute
`syslog-ng -s`

### Actual Behaviour

Got error:
```
Cannot load required module; module='http', details='The openobserve-log() driver depends on the syslog-ng http module, please install the syslog-ng-mod-http (Debian & derivatives) or the syslog-ng-http (RHEL & co) package', location='/usr/share/syslog-ng/include/scl/openobserve/openobserve.conf:44:2'
Error parsing config, syntax error, unexpected LL_ERROR, expecting '}' in /usr/share/syslog-ng/include/scl/openobserve/openobserve.conf:44:1-44:2:
39        body_suffix("]")
40        delimiter(",")
41        ...)
42      {
43
44----> @requires http "The openobserve-log() driver depends on the syslog-ng http module, please install the syslog-ng-mod-http (Debian & derivatives) or the syslog-ng-http (RHEL & co) package"
44----> ^
45
46          http(
47              url("`url`:`port`/api/`organization`/`stream`/_json")
48              user(`user`)
49              password(`password`)
```

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

BKPepe, 2025-06-19:
You need to enable zlib support in curl to have http enabled in syslog. I tried to explain it in https://github.com/openwrt/packages/commit/1e14d95d78d03ab163653166652972ca3e8c366e . By default, it is indeed currently disabled thus no http support.

skaa, 2025-06-19:
Thanks. And i found issue with same problem https://github.com/openwrt/packages/issues/20097. Closing this.

closed 2025-06-19 by skaa