openwrt/packages #26840: adblock-fast: Writes config file regularly

sigurasg, 2025-06-27:
### Package Name

adblock-fast

### Maintainer

Stan Grishin <stangri@melmac.ca>

### OpenWrt Version

24.10.2

### OpenWrt Target/Subtarget

ramips/mt7621

### Steps to Reproduce

1. Install adblock-fast.
2. Enable some blocklists.
3. Observe /etc/config/adblock-fast over time.
4. Notice that it is written regularly.

### Actual Behaviour

I'm noticing that the config file at `/etc/config/adblock-fast` is written regularly. Certainly on every boot for me, perhaps more often. It seems the service is updating the size option for each file URL.

I wonder if this is intentional, and if so whether it makes sense to provide an option not to do this to save on flash wear?
If this is transient information (as it would be for me, I don't persistently store the blocklists), it would be better to store this in /var/run (aka /tmp) somewhere?

This is from my config backup git repo, just showing how two random versions of the file differ.
```
$ git diff HEAD^ adblock-fast
diff --git a/router/etc/config/adblock-fast b/router/etc/config/adblock-fast
index 5fdbdb1..f33765b 100644
--- a/router/etc/config/adblock-fast
+++ b/router/etc/config/adblock-fast
@@ -35,22 +35,22 @@ config adblock-fast 'config'
 
 config file_url
        option url 'https://cdn.jsdelivr.net/gh/StevenBlack/hosts/hosts'
-       option size '5268235'
+       option size '6594233'
        option action 'block'
 
 config file_url
        option url 'https://raw.githubusercontent.com/AdguardTeam/cname-trackers/master/data/combined_disguised_trackers_justdomains.txt'
-       option size '3902071'
+       option size '3988053'
        option action 'block'
...
```

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

stangri, 2025-06-28:
Cannot reproduce the step 4. The config file is only updated with the new size values when either:
1. download/force redownload happen
2. size command is invoked

I'll see how quickly I can implement suggestion on not to update the file sizes in config. If/when it happens, it will be implemented in upstream repo first.

sigurasg, 2025-07-02:
So far I only see writes at boot time.

```
root@OpenWrt:~# uptime
 08:10:03 up 5 days, 37 min,  load average: 0.00, 0.01, 0.00
root@OpenWrt:~# date
Wed Jul  2 08:10:05 EDT 2025
root@OpenWrt:~# ls -ltc /etc/config
-rw-------    1 root     root          4425 Jun 27 07:54 dhcp
-rw-------    1 root     root          3224 Jun 27 07:36 adblock-fast
-rw-------    1 root     root           483 Dec 31  1969 acme
```

stangri, 2025-07-23:
@sigurasg please test https://dev.melmac.ca/repo/adblock-fast_1.1.4-r1_all.ipk with `uci set adblock-fast.config.update_config_sizes=0`.

sigurasg, 2025-07-24:
> [@sigurasg](https://github.com/sigurasg) please test https://dev.melmac.ca/repo/adblock-fast_1.1.4-r1_all.ipk with `uci set adblock-fast.config.update_config_sizes=0`.

Wow, thanks so much!
Sadly I'm travelling abroad and won't be able to test this until I get back, which is at least a couple of weeks out.

sigurasg, 2025-08-11:
I verify that `/etc/config/adblock-fast` is no longer modified on startup with `uci set adblock-fast.config.update_config_sizes=0`.
However, it appears `/etc/config/dhcp` is still written on each reboot?

```
root@OpenWrt:~# ls -ltc /etc/config/
-rw-------    1 root     root          4425 Aug 10 07:28 dhcp
-rw-------    1 root     root          3255 Aug 10 07:26 adblock-fast
-rw-------    1 root     root           483 Dec 31  1969 acme
```

sigurasg, 2025-08-13:
> Depends on the resolver option.

Interesting. I'm just using the default `dnsmasq`, I see there's support for `smartdns` and `unbound`. Maybe I should switch - I'm not seeing relative advantages to each in the READMEs. What's the lore?

stangri, 2025-08-13:
The differences may not be laid out in a comparison table, but they are described at https://docs.openwrt.melmac.ca/adblock-fast/#dns-resolver-option.

Some of those options need to be configured within the resolver's config file (ie: /etc/config/dhcp for dnsmasq), hence the updates.