openwrt/packages #26886: kea: dhcp4 server starts too early, needs to be restarted to actually function

nmeyerhans, 2025-07-03:
### Package Name

kea

### Maintainer

@rosysong

### OpenWrt Version

23.05

### OpenWrt Target/Subtarget

mvebu/cortexa9

### Steps to Reproduce

1. Configure an OpenWRT router to use kea as its DHCP service (IPv4). 
2. Boot the router
3. Attempt to obtain a DHCP lease from the router

Step 3 fails, as kea is not responding. Restarting kea resolves the problem.

Note that as this appears to be timing related, it may depend on the router having multiple internal interfaces, with kea configured to serve on each of them.

### Actual Behaviour

With an OpenWRT router configured with kea for DHCPv4 service, DHCP is unavailable following a reboot. The kea process must be restarted in order to properly handle requests. The following messages are logged by kea:

```
Wed Jun 11 16:43:23 2025 daemon.info kea-dhcp4[2050]: 2025-06-11 16:43:23.701 WARN  [kea-dhcp4.dhcpsrv/2050.3070166124] DHCPSRV_OPEN_SOCKET_FAIL failed to open socket: the interface br-xyz is not running
Wed Jun 11 16:43:23 2025 daemon.info kea-dhcp4[2050]: 2025-06-11 16:43:23.701 WARN  [kea-dhcp4.dhcpsrv/2050.3070166124] DHCPSRV_OPEN_SOCKET_FAIL failed to open socket: the interface br-abc is not running
Wed Jun 11 16:43:23 2025 daemon.info kea-dhcp4[2050]: 2025-06-11 16:43:23.740 INFO  [kea-dhcp4.dhcp4/2050.3070166124] DHCP4_OPEN_SOCKETS_FAILED maximum number of open service sockets attempts: 0, has been exhausted without success
Wed Jun 11 16:43:23 2025 daemon.info kea-dhcp4[2050]: 2025-06-11 16:43:23.741 WARN  [kea-dhcp4.dhcp4/2050.3070166124] DHCP4_MULTI_THREADING_INFO enabled: no, number of threads: 0, queue size: 0
Wed Jun 11 16:43:23 2025 daemon.info kea-dhcp4[2050]: 2025-06-11 16:43:23.741 INFO  [kea-dhcp4.dhcp4/2050.3070166124] DHCP4_STARTED Kea DHCPv4 server version 2.2.0 started
```

It seems as though kea is starting too soon in the boot process. Restarting the kea service once the interface configuration completes resolves the problem, and DHCP service is available as expected.

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

liuyuf78fk, 2025-07-17:
I also encountered the same issue: kea-dhcp4 starts too early and doesn't function until manually restarted.

I solved it by modifying the init script and adding a simple wrapper to introduce a delay during startup. This gives time for the network interfaces to fully initialize.

#### `/etc/init.d/kea-dhcp4`
```sh
START=98
STOP=60
USE_PROCD=1

NAME=kea-dhcp4
PROG=/usr/sbin/kea-dhcp4
WRAPPER=/usr/sbin/kea-dhcp4-wrapper
CONF=/etc/kea/kea-dhcp4.conf
LEASE_FILE=/var/lib/kea/kea-leases4.csv
LOGDIR=/var/log/kea
RUNDIR=/var/run/kea
START_DELAY=2

start_service() {
    mkdir -p $RUNDIR $LOGDIR /var/lib/kea
    [ -f "$LEASE_FILE" ] || touch "$LEASE_FILE"

    chmod 750 $RUNDIR
    chmod 750 $LOGDIR
    chmod 644 $LEASE_FILE
    chown root:root $RUNDIR $LOGDIR $LEASE_FILE

    procd_open_instance
    procd_set_param command "$WRAPPER" "$PROG" "$CONF" "$START_DELAY"
    procd_set_param stdout 1
    procd_set_param stderr 1
    procd_set_param respawn
    procd_close_instance
}
```

#### `/usr/sbin/kea-dhcp4-wrapper`
```sh
#!/bin/sh

BINARY="$1"
CONFIG="$2"
DELAY="${3:-0}"

if [ -z "$BINARY" ] || [ -z "$CONFIG" ]; then
    echo "Usage: $0 <binary> <config_file> [delay_seconds]" >&2
    exit 1
fi

[ "$DELAY" -gt 0 ] 2>/dev/null && sleep "$DELAY"

exec "$BINARY" -c "$CONFIG"
```

Krakonos, 2025-10-02:
Hi!

I have just hit the issue as well. My reason for using kea is the high availability mode (that so far seems to work great btw.!).

I dislike the solution above, mostly because of the arbitrary timeout, so I went to search for a better solution.

Reading the dnsmasq scripts and procd wiki, I found it. Mostly. I edited the init script in this way (see comments ADDED:):

```
#!/bin/sh /etc/rc.common

USE_PROCD=1
START=25
STOP=85

BIN_PATH="/usr/sbin"
CONF_PATH="/etc/kea"

start_service() {
	mkdir -p /var/run/kea

	config_load "kea"
	config_foreach start_kea "service"
}

start_kea() {
	local cfg="$1"

	config_get_bool disabled "$cfg" disabled 0
	[ "$disabled" = "0" ] || return

	config_get name "$cfg" name "$cfg"

	case "$name" in
		ctrl_agent|dhcp4|dhcp6|dhcp_ddns)
			name="${name/_/-}"
			cmd="${BIN_PATH}/kea-${name}"
			cnf="${CONF_PATH}/kea-${name}.conf"
			;;
		*)
			return 0
	esac

	procd_open_instance "$name"
	procd_set_param command "$cmd" -c "$cnf"
	procd_set_param env KEA_LOCKFILE_DIR=/tmp
	procd_append_param env KEA_PIDFILE_DIR=/tmp
	procd_set_param file "$cnf"
	procd_set_param stderr 1
	procd_set_param stdout 1
	procd_set_param netdev "lan_user" # <---- ADDED: netdev was added
	procd_close_instance
}

# ADDED: The triggers were added: ----->
service_triggers() {
	procd_add_reload_trigger "dhcp" "system"
	procd_add_interface_trigger "interface.*" "lan_user" /etc/init.d/kea restart
}
```

Obviously, I have hardcoded the interface name here, but the approach seems to work just fine. In addition, it works after interface reconfiguration, unplugging etc.

Now, I'm tempted to try "*" in there to see if it will work. However, this is still suboptimal in most cases. Dnsmasq adds trigger for each managed interface, but kea uci is currently... not using much.

Btw. I found that while wiki ( https://openwrt.org/docs/guide-developer/procd-init-scripts ) says the netdev parameter should be unix interface name, I found this not to be true - in fact, it seems to require the interface name as specified in "config interface XXXX". Not sure if this is the intent here, but I decided not to dig too deep into that.

I'd be willing to contribute a few bits and pieces, but I'm unsure what approach would be wanted? For example:

* We could add 'use_kea' option in the /etc/config to setup the interfaces.
* We could add interface config in the /etc/config/kea

... but I personally don't have the capacity to add full kea config support -- there are a lot of options and only being able to generate a subset would be not that useful. Even my basic config has a few extra options.

Thoughts?

PS: I'd love if openwrt moved from dnsmasq to kea + unbound!