openwrt/packages #26897: nginx: Missing in the latest snapshot APK package feed

HsuryX, 2025-07-05:
### Package Name

nginx

### Maintainer

@drujd

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

mediatek/filogic

### Steps to Reproduce

Run `apk update && apk list nginx` and see the return list, cannot find a lot of nginx related packages in the latest [snapshot APK package feed](https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/), e.g. nginx and nginx-mod-luci.

I noticed this because failing to build the image today, but it worked last week:
```
...
17.46 ERROR: unable to select packages:
17.46   nginx (no such package):
17.46     required by: luci-nginx-24.298.47280~0535c07[nginx]
17.46   nginx-mod-luci (no such package):
17.46     required by: luci-nginx-24.298.47280~0535c07[nginx-mod-luci]
...
```

### Actual Behaviour

Not sure, maybe something about nginx failed within the OpenWRT package CI/CD pipeline?

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

HsuryX, 2025-07-06:
Seems to be related to PR https://github.com/openwrt/packages/pull/26889

drujd, 2025-07-06:
FYI I'm not a maintainer of the nginx packages. However, given that nothing changed in the package itself (makefile etc.) recently, it seems it has nothing to do with the actual nginx package.

Previous comment migjt be on point.