openwrt/packages #27011: Does openlist/alist depend on fuse?

jjm2473, 2025-07-16:
https://github.com/openwrt/packages/blob/978dfa5be02c2e4593f911a2f201ea58d0eb10fd/net/openlist/Makefile#L40

According to the openlist documentation https://docs.oplist.org/ , there is no mention of any support for mounting, so why it depends on fuse?

@1715173329

jjm2473, 2025-07-16:
This is just a placeholder. It existed three years ago and has not been implemented yet.

https://github.com/OpenListTeam/OpenList/blob/main/internal/fuse/fs.go#L10-L23
```go
func (fs *Fs) Init() {
	//TODO implement me
	panic("implement me")
}

func (fs *Fs) Destroy() {
	//TODO implement me
	panic("implement me")
}

func (fs *Fs) Statfs(path string, stat *fuse.Statfs_t) int {
	//TODO implement me
	panic("implement me")
}
```

At present, it should not be necessary to depends on fuse.

1715173329, 2025-07-16:
mount.go implements a different function.
I don't see a user of fuse package in the code though. Probably fine to remove.

closed 2025-08-01 by 1715173329