openwrt/packages #27014: luajit2: cross-build on macOS does not work

BKPepe, 2025-07-16:
### Package Name

luajit2

### Maintainer

@jmarcet

### OpenWrt Version

snapshot

### OpenWrt Target/Subtarget

mvebu/cortex-a9

### Steps to Reproduce

1. Have macOS
2. Try to compile luajit2

### Actual Behaviour

Right now, it fails:
==== Building LuaJIT 2.1 ====
make -C src
make[4]: Entering directory '/Volumes/OpenWrt/omnia-hbd/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/luajit2-2.1.2024.08.15/src'
HOSTCC    host/minilua.o
HOSTLINK  host/minilua
VERSION   luajit.h
DYNASM    host/buildvm_arch.h
HOSTCC    host/buildvm.o
HOSTCC    host/buildvm_asm.o
HOSTCC    host/buildvm_peobj.o
HOSTCC    host/buildvm_lib.o
HOSTCC    host/buildvm_fold.o
HOSTLINK  host/buildvm
BUILDVM   lj_vm.S
Error: pointer size mismatch in cross-build.
Try: make HOST_CC="gcc -m32" CROSS=...

make[4]: *** [Makefile:695: lj_vm.S] Error 1

maybe: @httpstorm can help with this one. :-)

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

httpstorm, 2025-07-17:
I think it is trying to compile the host architecture instead of the target. Or it gets confused by the `-m32` flag.
This looks like missing configuration, but I don't have much experience with the OpenWRT make system.
I'll check the assembly later, meanwhile here is my progress:

``` C
make package/feeds/packages/luajit2/{clean,compile} -j 1 V=s
…
==== Building LuaJIT 2.1 ====
make -C src
make[4]: Entering directory '/Volumes/lua/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/luajit2-2.1.2024.11.13/src'
HOSTCC    host/minilua.o
HOSTLINK  host/minilua
ld: warning: The i386 architecture is deprecated for macOS (remove from the Xcode build setting: ARCHS)
ld: warning: ignoring file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/lib/libm.tbd, missing required architecture i386 in file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/lib/libm.tbd (3 slices)
ld: warning: ignoring file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/lib/libSystem.tbd, missing required architecture i386 in file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/lib/libSystem.tbd (3 slices)
Undefined symbols for architecture i386:
…
// these are common symbols provided by the C/C++ runtime library:
// ___bzero ___error _malloc _memcmp _memcpy _strlen _strncat _strncpy _strtoul …
…
ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Build logs from: macOS 15.5 Intel, Xcode 16.4
[mvebu / cortex-a9 / WRT3200ACM](https://httpstorm.com/openwrt/test/2025-07-17%20postgresql/lua-mvebu.cortex-a9.txt)
[Freescale MPC85xx / P1010 / TL-WDR4900 v1](https://httpstorm.com/openwrt/test/2025-07-17%20postgresql/lua-openwrt-24.10-powerpc-8548.txt)

Attempting to build the host tools
``` bash
make package/feeds/packages/luajit2/host/{clean,compile} -j 1 V=s
…
Makefile:326: *** missing: export MACOSX_DEPLOYMENT_TARGET=XX.YY.  Stop.
```

If I add this to `Host/Compile`, the error is resolved, but the package still fails.
``` bash
MACOSX_DEPLOYMENT_TARGET="15.5"
```

If I remove `HOST_BITS := -m32` or change it to `HOST_BITS := -m64`, I reproduce your error:
``` bash
BUILDVM   lj_vm.S
Error: pointer size mismatch in cross-build.
Try: make HOST_CC="gcc -m32" CROSS=...
```

httpstorm, 2025-07-18:
I did try updating to 2025-05-29 before I made my last comment, but it makes no difference, so I didn't mention it.

httpstorm, 2025-07-18:
This can be added conditionally to `Host/Compile` to get the host version automatically, but only for macOS hosts:
``` makefile
MACOSX_DEPLOYMENT_TARGET="$(shell sw_vers --productVersion)"

# with Darwin detection
MACOSX_DEPLOYMENT_TARGET="$(shell uname -s | grep -q Darwin && sw_vers --productVersion)"
```

I also got this error while building for the host
``` bash
make package/feeds/packages/luajit2/host/{clean,compile} -j 1 V=s
…
make[3]: Leaving directory '/Volumes/lua/openwrt/build_dir/hostpkg/luajit2-2.1.2025.05.29'
cp -fpR /Volumes/lua/openwrt/staging_dir/hostpkg/bin/luajit-* /Volumes/lua/openwrt/staging_dir/hostpkg/bin/luajit2
cp: target '/Volumes/lua/openwrt/staging_dir/hostpkg/bin/luajit2': Not a directory

# resolved by
cd staging_dir/hostpkg/bin/
rm luajit luajit2 luajit-*
```

mirko, 2026-09-30:
There is a vibe-coded approach to get rid of all the host-target pointer mismatch issues with luajit2: https://github.com/openwrt/packages/pull/30658. While it might be rightfully called out as AI slop, the result is byte-identical to the current workarounds, which don't even work in scenarios like this very macos-one.