openwrt/packages #27025: knot-resolver: lack some optional dependencies in submenu

CoelacanthusHex, 2025-07-18:
### Package Name

knot-resolver

### Maintainer

Jan Pavlinec jan.pavlinec@nic.cz

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

mediatek/filogic

### Steps to Reproduce

1. Install knot-resolver.
2. Try to use some features like policy.slice_randomize_psl(), policy.rpz(), http, dns64, experimental_dot_auth and prefill modules
3. You will fail to start knot-resolver because lack lua libraries.

### Actual Behaviour

You will fail to start knot-resolver because lack lua libraries if try to use some features like policy.slice_randomize_psl(), policy.rpz(), http, dns64, experimental_dot_auth and prefill modules.
There is some optional Lua libraries for knot-resolver:
* lua-basexx: experimental_dot_auth module
* lua-cqueues: http and dns64 module, policy.rpz() function
* lua-http: http and prefill modules, trust_anchors bootstrap
* lua-psl: policy.slice_randomize_psl() function
Since they are not necessary, maybe we can add options for them and place it in submenu.

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

BKPepe, 2025-07-27:
cc: @vcunat same thing is definitely in Turris OS

vcunat, 2025-07-28:
Yes, I believe noone has packaged these lua libraries for OpenWrt.

vcunat, 2025-07-28:
The list might not be precise, though.  I believe that `dns64` doesn't need anything.  RPZ only needs it for auto-reloading (which seems minor).  And I honestly don't get why someone would even want experimental_dot_auth or trust anchor bootstrap.