openwrt/packages #27039: rclone: ftp tls handshake fail - no shared cipher

starmoonxp, 2025-07-20:
### Package Name

rclone

### Maintainer

（@Elon Huang)

### OpenWrt Version

24.10.2

### OpenWrt Target/Subtarget

mediatek / filogic 

### Steps to Reproduce

root@OpenWrt:~# rclone.1.70.2 --version
rclone v1.70.2
- os/version: openwrt 24.10.2 (64 bit)
- os/kernel: 6.6.93 (aarch64)
- os/type: linux
- os/arch: arm64 (ARMv8 compatible)
- go/version: go1.23.10
- go/linking: dynamic
- go/tags: none

root@OpenWrt:~# rclone.1.70.2 -vv lsf starmoon: --config=/etc/rclone/rclone.
conf
2025/07/20 18:52:28 DEBUG : rclone: Version "v1.70.2" starting with parameters ["rclone.1.70.2" "-vv" "lsf" "starmoon:" "--config=/etc/rclone/rclone.conf"]
2025/07/20 18:52:28 DEBUG : Creating backend with remote "starmoon:"
2025/07/20 18:52:28 DEBUG : Using config file from "/etc/rclone/rclone.conf"
2025/07/20 18:52:28 DEBUG : ftp://<starmoon_ftp>:21: Connecting to FTP server
2025/07/20 18:52:28 DEBUG : ftp://<starmoon_ftp>:21: dial("tcp","<starmoon_ftp>:21")
2025/07/20 18:52:28 DEBUG : ftp://<starmoon_ftp>:21: > dial: conn=117.84.102.242:54996->121.235.237.101:21, err=<nil>
2025/07/20 18:52:28 CRITICAL: Failed to create file system for "starmoon:": NewFs: failed to make FTP connection to "starmoon.org.cn:21": remote error: tls: handshake failure

==> /var/log/proftpd/proftpd.log <==
Jul 21 02:16:17 linux.localdomain proftpd[9032] 192.168.18.1 (::ffff:117.84.102.242[::ffff:117.84.102.242]): FTP session closed.
Jul 21 02:16:29 linux.localdomain proftpd[9202] 192.168.18.1 (::ffff:117.84.102.242[::ffff:117.84.102.242]): FTP session opened.

==> /var/log/proftpd/tls.log <==
Jul 21 02:16:29 mod_tls/2.4.2[9202]: TLS/TLS-C requested, starting TLS handshake
Jul 21 02:16:29 mod_tls/2.4.2[9202]: unable to accept TLS connection: protocol error:
  (1) error:1408A0C1:SSL routines:SSL3_GET_CLIENT_HELLO:no shared cipher
Jul 21 02:16:29 mod_tls/2.4.2[9202]: TLS/TLS-C negotiation failed on control channel

==> /var/log/proftpd/proftpd.log <==
Jul 21 02:16:29 linux.localdomain proftpd[9202] 192.168.18.1 (::ffff:117.84.102.242[::ffff:117.84.102.242]): FTP session closed.

================================================
root@OpenWrt:~# rclone.1.69.1  --version
rclone v1.69.1
- os/version: openwrt 24.10.2 (64 bit)
- os/kernel: 6.6.93 (aarch64)
- os/type: linux
- os/arch: arm64 (ARMv8 compatible)
- go/version: go1.21.13
- go/linking: dynamic
- go/tags: none
root@OpenWrt:~# rclone.1.69.1  -vv lsf starmoon: --config=/etc/rclone/rclone
.conf
2025/07/20 18:57:03 DEBUG : rclone: Version "v1.69.1" starting with parameters ["rclone.1.69.1" "-vv" "lsf" "starmoon:" "--config=/etc/rclone/rclone.conf"]
2025/07/20 18:57:03 DEBUG : Creating backend with remote "starmoon:"
2025/07/20 18:57:03 DEBUG : Using config file from "/etc/rclone/rclone.conf"
2025/07/20 18:57:03 DEBUG : ftp://<starmoon_ftp>:21: Connecting to FTP server
2025/07/20 18:57:03 DEBUG : ftp://<starmoon_ftp>:21: dial("tcp","<starmoon_ftp>:21")
2025/07/20 18:57:03 DEBUG : ftp://<starmoon_ftp>n:21: > dial: conn=*fshttp.timeoutConn, err=<nil>
2025/07/20 18:57:03 DEBUG : ftp://<starmoon_ftp>:21: dial("tcp","121.235.237.101:60138")
2025/07/20 18:57:03 DEBUG : ftp://<starmoon_ftp>n:21: > dial: conn=*tls.Conn, err=<nil>
movie/
2025/07/20 18:57:03 DEBUG : 3 go routines active
2025/07/20 18:57:03 DEBUG : ftp://<starmoon_ftp>:21: closing 1 unused connections

==> /var/log/proftpd/proftpd.log <==
Jul 21 02:57:03 linux.localdomain proftpd[17167] 192.168.18.1 (::ffff:117.84.102.242[::ffff:117.84.102.242]): FTP session opened.

==> /var/log/proftpd/tls.log <==
Jul 21 02:57:03 mod_tls/2.4.2[17167]: TLS/TLS-C requested, starting TLS handshake
Jul 21 02:57:03 mod_tls/2.4.2[17167]: TLSv1/SSLv3 connection accepted, using cipher AES128-GCM-SHA256 (128 bits)

==> /var/log/proftpd/proftpd.log <==
Jul 20 18:57:03 linux.localdomain proftpd[17167] 192.168.18.1 (::ffff:117.84.102.242[::ffff:117.84.102.242]): Preparing to chroot to directory '/home/share'
Jul 20 18:57:03 linux.localdomain proftpd[17167] 192.168.18.1 (::ffff:117.84.102.242[::ffff:117.84.102.242]): USER movierc: Login successful.

### Actual Behaviour

It works fine with rclone 1.69.1 but failed when I upgraded to 1.70.2 with same configuration file(s). 

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

starmoonxp, 2025-07-25:
> Please report this issue to upstream rclone.

Upstream rclone confirmed this "bug" and made a fix for this. But only in beta versions (1.71.0-beta.8961 or later).

1715173329, 2025-07-29:
please check #27114