openwrt/packages #27085: atop: package build fails

risa2000, 2025-07-25:
### Package Name

atop

### Maintainer

(@hnyman, @robimarko)

### OpenWrt Version

current master (f931ddb99f32c495620cf11e9d2eaf3c1ab9f5bb)q

### OpenWrt Target/Subtarget

ramips/mt7621

### Steps to Reproduce

Building the current master branch fails at building `atop-2.11.0` with error:

```
make[3]: Entering directory '/home/risa/work/openwrt-build-env/openwrt/feeds/packages/admin/atop'
rm -f /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/atop-2.11.0/.built
touch /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/atop-2.11.0/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float   -fmacro-prefix-map=/home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/atop-2.11.0=atop-2.11.0 -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-15.1.0_musl/usr/include -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-15.1.0_musl/include -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-15.1.0_musl/include/fortify     " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float   -fmacro-prefix-map=/home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/atop-2.11.0=atop-2.11.0 -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -mips16 -minterlink-mips16  -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-15.1.0_musl/usr/include -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-15.1.0_musl/include -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-15.1.0_musl/include/fortify     " LDFLAGS="-L/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-15.1.0_musl/usr/lib -L/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-15.1.0_musl/lib -fuse-ld=bfd   -znow -zrelro   " make -j1 -C /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/atop-2.11.0/. AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/atop-2.11.0=atop-2.11.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="mipsel-openwrt-linux-musl-ld.bfd" NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CROSS="mipsel-openwrt-linux-musl-" ARCH="mipsel" ;
make[4]: Entering directory '/home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/atop-2.11.0'
mipsel-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float   -fmacro-prefix-map=/home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/atop-2.11.0=atop-2.11.0 -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-15.1.0_musl/usr/include -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-15.1.0_musl/include -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-15.1.0_musl/include/fortify      -O2 -I. -Wall -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/glib-2.0 -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/glib-2.0/include -I/home/risa/work/openwrt-build-env/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include  -Wno-stringop-truncation -Wmissing-prototypes -Wmissing-declarations -Wformat-security    -c -o showprocs.o showprocs.c
showprocs.c: In function 'showprocline':
showprocs.c:500:38: error: too many arguments to function 'curelem.pf->doactiveconvert'; expected 0, have 3
  500 |                         printg("%s", curelem.pf->doactiveconvert(curstat,
      |                                      ^~~~~~~                     ~~~~~~~
In file included from showprocs.c:61:
showlinux.h:105:17: note: declared here
  105 |         char *(*doactiveconvert)();      // pointer to conv function
      |                 ^~~~~~~~~~~~~~~
showprocs.c:505:38: error: too many arguments to function 'curelem.pf->doexitconvert'; expected 0, have 3
  505 |                         printg("%s", curelem.pf->doexitconvert(curstat,
      |                                      ^~~~~~~                   ~~~~~~~
showlinux.h:107:17: note: declared here
  107 |         char *(*doexitconvert)();        // pointer to conv function
      |                 ^~~~~~~~~~~~~
showprocs.c: At top level:
showprocs.c:582:20: error: initialization of 'char * (*)(void)' from incompatible pointer type 'char * (*)(struct tstat *, int,  int)' [-Wincompatible-pointer-types]
  582 |    { "TID", "TID", procprt_TID_ae, procprt_TID_ae, ' ', 5}; //DYNAMIC WIDTH!
      |                    ^~~~~~~~~~~~~~
showprocs.c:582:20: note: (near initialization for 'procprt_TID.doactiveconvert')
showprocs.c:570:1: note: 'procprt_TID_ae' declared here
  570 | procprt_TID_ae(struct tstat *curstat, int avgval, int nsecs)
      | ^~~~~~~~~~~~~~
showprocs.c:582:36: error: initialization of 'char * (*)(void)' from incompatible pointer type 'char * (*)(struct tstat *, int,  int)' [-Wincompatible-pointer-types]
  582 |    { "TID", "TID", procprt_TID_ae, procprt_TID_ae, ' ', 5}; //DYNAMIC WIDTH!
      |                                    ^~~~~~~~~~~~~~
showprocs.c:582:36: note: (near initialization for 'procprt_TID.doexitconvert')
showprocs.c:570:1: note: 'procprt_TID_ae' declared here

... removed 1100 line ...

make[4]: *** [<builtin>: showprocs.o] Error 1
make[4]: Leaving directory '/home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/atop-2.11.0'
make[3]: *** [Makefile:57: /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/atop-2.11.0/.built] Error 2
make[3]: Leaving directory '/home/risa/work/openwrt-build-env/openwrt/feeds/packages/admin/atop'
time: package/feeds/packages/atop/compile#0.06#0.00#0.07
    ERROR: package/feeds/packages/atop failed to build.
make[2]: *** [package/Makefile:183: package/feeds/packages/atop/compile] Error 1
```

### Actual Behaviour

Build failed.

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2025-07-26:
Please post your diffconfig, as default config does not break like this

risa2000, 2025-07-27:
@robimarko Here you go.

[config.diff.txt](https://github.com/user-attachments/files/21453764/config.diff.txt)

hnyman, 2025-07-27:
> ```
> # CONFIG_GCC_USE_VERSION_14 is not set
> CONFIG_GCC_USE_VERSION_15=y
> CONFIG_GCC_VERSION="15.1.0"
> CONFIG_GCC_VERSION_15=y
> ```

First guess would be the non-standard GCC 15...
Several packages have required compatibility fixes, see the repository commits regarding fixes and try to figure out if that works for you,

https://github.com/search?q=repo%3Aopenwrt%2Fpackages+gcc+15&type=commits&s=committer-date&o=desc

robimarko, 2025-07-27:
Updating to 2.12.0 fixes this or setting GNU17 standard

risa2000, 2025-07-27:
@robimarko I can confirm that bumping to `atop-2.12.0` fixed the build problem.

PR #27101 (2025-07-27): atop: update to 2.12.0
## 📦 Package Details

**Maintainer:** @robimarko 

2.11.0 will not compile with GCC15, so update to 2.12.0. It also fixes CVE-2025-31160.

Fixes: #27085
---

## 🧪 Run Testing Details

- **OpenWrt Version:** main
- **OpenWrt Target/Subtarget:** ramips/mt7621
- **OpenWrt Device:** no

---

## ✅ Formalities

- [x] I have reviewed the [CONTRIBUTING.md](../CONTRIBUTING.md) file for detailed contributing guidelines.

### If your PR contains a patch:

- [ ] It can be applied using `git am`
- [ ] It has been refreshed to avoid offsets, fuzzes, etc., using
  ```bash
  make packag [... cut]
closed 2025-07-27 by risa2000