openwrt/packages #27102: bind: <short description>

peterwillcn, 2025-07-28:
### Package Name

bind

### Maintainer

@Noah Meyerhans

### OpenWrt Version

snapshot

### OpenWrt Target/Subtarget

rockchip

### Steps to Reproduce

make[3]: Leaving directory '/workdir/package/feeds/net/bind' time: package/feeds/net/bind/compile#24.62#3.07#29.10 [233/266] Building C object libarchive/CMakeFiles/archive_static.dir/archive_write_set_format_shar.c.o ERROR: package/feeds/net/bind failed to build. make[2]: *** [package/Makefile:183: package/feeds/net/bind/compile] Error 1

### Actual Behaviour

make[3]: Leaving directory '/workdir/package/feeds/net/bind' time: package/feeds/net/bind/compile#24.62#3.07#29.10 [233/266] Building C object libarchive/CMakeFiles/archive_static.dir/archive_write_set_format_shar.c.o ERROR: package/feeds/net/bind failed to build. make[2]: *** [package/Makefile:183: package/feeds/net/bind/compile] Error 1

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Alphix, 2025-07-28:
@peterwillcn see #27095

peterwillcn, 2025-07-28:
@robimarko ping

Alphix, 2025-07-28:
> Building C object libarchive/CMakeFiles/archive_static.dir/archive_write_set_format_shar.c.o

Wait...that's not in `bind`, though?

vochong, 2025-07-31:
@Alphix 

I've got this bind compile error.   Please take a look.  Thanks!

```
make[3]: Entering directory '/home/runner/work/qosmio-nss/qosmio-nss/feeds/packages/net/bind'
rm -f /home/runner/work/qosmio-nss/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/bind-9.20.11/.configured_*
rm -f /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/stamp/.bind_installed
(cd /home/runner/work/qosmio-nss/qosmio-nss/build_dir/target-aarch64_cortex-a53_musl/bind-9.20.11; rm -f aclocal.m4; rm -f libtool.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf ./autom4te.cache; [ -e ./config.rpath ] || ln -s /home/runner/work/qosmio-nss/qosmio-nss/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/autom4te AUTOCONF=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/autoconf AUTOMAKE=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/automake ACLOCAL=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/aclocal AUTOHEADER=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/libtoolize LIBTOOL=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/libtool M4=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/m4 AUTOPOINT=true GTKDOCIZE=true LIBTOOLIZE='/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/libtoolize --install' /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/autoreconf -v -f -i -B /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/share/aclocal -I /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/host/share/aclocal -I /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/hostpkg/share/aclocal -I /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/usr/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/aclocal -I /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/host/share/aclocal -I /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/hostpkg/share/aclocal -I /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/usr/share/aclocal -I m4 -I . --force 
fatal: not a git repository (or any of the parent directories): .git
autoreconf: configure.ac: tracing
fatal: not a git repository (or any of the parent directories): .git
autoreconf: running: /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/libtoolize --install --copy --force
OpenWrt-libtoolize: putting auxiliary files in '.'.
OpenWrt-libtoolize: copying file './config.guess'
OpenWrt-libtoolize: copying file './config.sub'
OpenWrt-libtoolize: copying file './install-sh'
OpenWrt-libtoolize: copying file './ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
OpenWrt-libtoolize: copying file 'm4/libtool.m4'
OpenWrt-libtoolize: copying file 'm4/ltoptions.m4'
OpenWrt-libtoolize: copying file 'm4/ltsugar.m4'
OpenWrt-libtoolize: copying file 'm4/ltversion.m4'
OpenWrt-libtoolize: copying file 'm4/lt~obsolete.m4'
OpenWrt-libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/aclocal -I /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/host/share/aclocal -I /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/hostpkg/share/aclocal -I /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/usr/share/aclocal -I m4 -I . --force 
fatal: not a git repository (or any of the parent directories): .git
autoreconf: running: /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/autoconf --include=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/host/share/aclocal --include=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/hostpkg/share/aclocal --include=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/share/aclocal --force
fatal: not a git repository (or any of the parent directories): .git
autoreconf: running: /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/autoheader --include=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/host/share/aclocal --include=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/hostpkg/share/aclocal --include=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/share/aclocal --force
fatal: not a git repository (or any of the parent directories): .git
autoreconf: running: /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/host/bin/automake --add-missing --copy --force-missing
fatal: not a git repository (or any of the parent directories): .git
configure.ac:69: installing './compile'
configure.ac:51: installing './missing'
[... cut ...]
make[2]: *** [package/Makefile:183: package/feeds/packages/bind/compile] Error 1
make[2]: Leaving directory '/home/runner/work/qosmio-nss/qosmio-nss'
make[1]: *** [package/Makefile:177: /home/runner/work/qosmio-nss/qosmio-nss/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/qosmio-nss/qosmio-nss'
make: *** [/home/runner/work/qosmio-nss/qosmio-nss/include/toplevel.mk:233: world] Error 2
Error: Process completed with exit code 2.
```

Alphix, 2025-07-31:
> [@Alphix](https://github.com/Alphix)
> 
> I've got this bind compile error. Please take a look. Thanks!
> 

@vochong: I've already linked https://github.com/openwrt/packages/pull/27095 above...it should fix this, did you apply those patches first?

vochong, 2025-07-31:
@Alphix 

No I did not apply the patches mentioned.  I will wait until your PR is committed and try again. Thanks!

closed 2025-09-14 by peterwillcn