openwrt/packages #27111: parted: does not compile on gcc15

pwned-pixel, 2025-07-29:
### Package Name

parted

### Maintainer

Oskari Rauta <oskari.rauta@gmail.com>

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

Mediatek/Filogic (MT798x)

### Steps to Reproduce

diffconfig:

```
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_openwrt_one=y
CONFIG_PACKAGE_ccrypt=y
CONFIG_PACKAGE_libblkid=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libparted=y
CONFIG_PACKAGE_libpthread=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libuuid=y
CONFIG_PACKAGE_parted=y
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PARTED_READLINE=y
```

### Actual Behaviour

error message:

```libtool: link: (cd ".libs" && rm -f "libparted.so.2" && ln -s "libparted.so.2.0.5" "libparted.so.2")
libtool: link: (cd ".libs" && rm -f "libparted.so" && ln -s "libparted.so.2.0.5" "libparted.so")
libtool: link: (cd .libs/libparted.lax/libfs.a && aarch64-openwrt-linux-musl-gcc-ar x "/home/user/nolf/openwrt/one/openwrt/build_dir/target-aarch64_cortex-a53_musl/parted-3.6/libparted/fs/.libs/libfs.a")
libtool: link: (cd .libs/libparted.lax/liblabels.a && aarch64-openwrt-linux-musl-gcc-ar x "/home/user/nolf/openwrt/one/openwrt/build_dir/target-aarch64_cortex-a53_musl/parted-3.6/libparted/labels/.libs/liblabels.a")
libtool: link: (cd .libs/libparted.lax/libgnulib.a && aarch64-openwrt-linux-musl-gcc-ar x "/home/user/nolf/openwrt/one/openwrt/build_dir/target-aarch64_cortex-a53_musl/parted-3.6/libparted/../lib/.libs/libgnulib.a")
libtool: link: aarch64-openwrt-linux-musl-gcc-ar cr .libs/libparted.a  debug.o architecture.o device.o exception.o filesys.o libparted.o timer.o unit.o disk.o cs/geom.o cs/constraint.o cs/natmath.o arch/linux.o  .libs/libparted.lax/libfs.a/a-interface.o .libs/libparted.lax/libfs.a/affs.o .libs/libparted.lax/libfs.a/amiga.o .libs/libparted.lax/libfs.a/apfs.o .libs/libparted.lax/libfs.a/asfs.o .libs/libparted.lax/libfs.a/bootsector.o .libs/libparted.lax/libfs.a/btrfs.o .libs/libparted.lax/libfs.a/f2fs.o .libs/libparted.lax/libfs.a/fat.o .libs/libparted.lax/libfs.a/hfs.o .libs/libparted.lax/libfs.a/interface.o .libs/libparted.lax/libfs.a/jfs.o .libs/libparted.lax/libfs.a/linux_swap.o .libs/libparted.lax/libfs.a/nilfs2.o .libs/libparted.lax/libfs.a/ntfs.o .libs/libparted.lax/libfs.a/probe.o .libs/libparted.lax/libfs.a/reiserfs.o .libs/libparted.lax/libfs.a/udf.o .libs/libparted.lax/libfs.a/ufs.o .libs/libparted.lax/libfs.a/xfs.o  .libs/libparted.lax/liblabels.a/aix.o .libs/libparted.lax/liblabels.a/atari.o .libs/libparted.lax/liblabels.a/bsd.o .libs/libparted.lax/liblabels.a/dos.o .libs/libparted.lax/liblabels.a/dvh.o .libs/libparted.lax/liblabels.a/efi_crc32.o .libs/libparted.lax/liblabels.a/gpt.o .libs/libparted.lax/liblabels.a/loop.o .libs/libparted.lax/liblabels.a/mac.o .libs/libparted.lax/liblabels.a/pc98.o .libs/libparted.lax/liblabels.a/pt-tools.o .libs/libparted.lax/liblabels.a/rdb.o .libs/libparted.lax/liblabels.a/sun.o  .libs/libparted.lax/libgnulib.a/libgnulib_la-argmatch.o .libs/libparted.lax/libgnulib.a/libgnulib_la-basename-lgpl.o .libs/libparted.lax/libgnulib.a/libgnulib_la-basename.o .libs/libparted.lax/libgnulib.a/libgnulib_la-c-ctype.o .libs/libparted.lax/libgnulib.a/libgnulib_la-c-strcasecmp.o .libs/libparted.lax/libgnulib.a/libgnulib_la-c-strncasecmp.o .libs/libparted.lax/libgnulib.a/libgnulib_la-canonicalize-lgpl.o .libs/libparted.lax/libgnulib.a/libgnulib_la-cloexec.o .libs/libparted.lax/libgnulib.a/libgnulib_la-close-stream.o .libs/libparted.lax/libgnulib.a/libgnulib_la-closeout.o .libs/libparted.lax/libgnulib.a/libgnulib_la-dirname-lgpl.o .libs/libparted.lax/libgnulib.a/libgnulib_la-dirname.o .libs/libparted.lax/libgnulib.a/libgnulib_la-dynarray_at_failure.o .libs/libparted.lax/libgnulib.a/libgnulib_la-dynarray_emplace_enlarge.o .libs/libparted.lax/libgnulib.a/libgnulib_la-dynarray_finalize.o .libs/libparted.lax/libgnulib.a/libgnulib_la-dynarray_resize.o .libs/libparted.lax/libgnulib.a/libgnulib_la-dynarray_resize_clear.o .libs/libparted.lax/libgnulib.a/libgnulib_la-error.o .libs/libparted.lax/libgnulib.a/libgnulib_la-exitfail.o .libs/libparted.lax/libgnulib.a/libgnulib_la-fcntl.o .libs/libparted.lax/libgnulib.a/libgnulib_la-fd-hook.o .libs/libparted.lax/libgnulib.a/libgnulib_la-free.o .libs/libparted.lax/libgnulib.a/libgnulib_la-getopt.o .libs/libparted.lax/libgnulib.a/libgnulib_la-getopt1.o .libs/libparted.lax/libgnulib.a/libgnulib_la-getprogname.o .libs/libparted.lax/libgnulib.a/libgnulib_la-hard-locale.o .libs/libparted.lax/libgnulib.a/libgnulib_la-ialloc.o .libs/libparted.lax/libgnulib.a/libgnulib_la-localcharset.o .libs/libparted.lax/libgnulib.a/libgnulib_la-lock.o .libs/libparted.lax/libgnulib.a/libgnulib_la-long-options.o .libs/libparted.lax/libgnulib.a/libgnulib_la-malloca.o .libs/libparted.lax/libgnulib.a/libgnulib_la-mbrtowc.o .libs/libparted.lax/libgnulib.a/libgnulib_la-nl_langinfo.o .libs/libparted.lax/libgnulib.a/libgnulib_la-progname.o .libs/libparted.lax/libgnulib.a/libgnulib_la-quotearg.o .libs/libparted.lax/libgnulib.a/libgnulib_la-rawmemchr.o .libs/libparted.lax/libgnulib.a/libgnulib_la-regex.o .libs/libparted.lax/libgnulib.a/libgnulib_la-rpmatch.o .libs/libparted.lax/libgnulib.a/libgnulib_la-safe-read.o .libs/libparted.lax/libgnulib.a/libgnulib_la-scratch_buffer_grow.o .libs/libparted.lax/libgnulib.a/libgnulib_la-scratch_buffer_grow_preserve.o .libs/libparted.lax/libgnulib.a/libgnulib_la-scratch_buffer_set_array_size.o .libs/libparted.lax/libgnulib.a/libgnulib_la-setlocale_null.o .libs/libparted.lax/libgnulib.a/libgnulib_la-stat-time.o .libs/libparted.lax/libgnulib.a/libgnulib_la-stripslash.o .libs/libparted.lax/libgnulib.a/libgnulib_la-strtoll.o .libs/libparted.lax/libgnulib.a/libgnulib_la-strtoull.o .libs/libparted.lax/libgnulib.a/libgnulib_la-tempname.o .libs/libparted.lax/libgnulib.a/libgnulib_la-threadlib.o .libs/libparted.lax/libgnulib.a/libgnulib_la-unistd.o .libs/libparted.lax/libgnulib.a/libgnulib_la-version-etc-fsf.o .libs/libparted.lax/libgnulib.a/libgnulib_la-version-etc.o .libs/libparted.lax/libgnulib.a/libgnulib_la-wctype-h.o .libs/libparted.lax/libgnulib.a/libgnulib_la-xalloc-die.o .libs/libparted.lax/libgnulib.a/libgnulib_la-xmalloc.o .libs/libparted.lax/libgnulib.a/libgnulib_la-xstrtol.o .libs/libparted.lax/libgnulib.a/libgnulib_la-xstrtoll.o .libs/libparted.lax/libgnulib.a/libgnulib_la-xstrtoul.o .libs/libparted.lax/libgnulib.a/libgnulib_la-xstrtoull.o 
libtool: link: aarch64-openwrt-linux-musl-gcc-ranlib .libs/libparted.a
libtool: link: rm -fr .libs/libparted.lax
libtool: link: ( cd ".libs" && rm -f "libparted.la" && ln -s "../libparted.la" "libparted.la" )
make[6]: Leaving directory '/home/user/nolf/openwrt/one/openwrt/build_dir/target-aarch64_cortex-a53_musl/parted-3.6/libparted'
make[5]: Leaving directory '/home/user/nolf/openwrt/one/openwrt/build_dir/target-aarch64_cortex-a53_musl/parted-3.6/libparted'
Making all in parted
make[5]: Entering directory '/home/user/nolf/openwrt/one/openwrt/build_dir/target-aarch64_cortex-a53_musl/parted-3.6/parted'
rm -f version.c
printf '#include <config.h>\n' > version.ct
printf 'char const *Version = "3.6";\n' >> version.ct
chmod a-w version.ct
mv version.ct version.c
rm -f version.h
printf 'extern char const *Version;\n' > version.ht
chmod a-w version.ht
mv version.ht version.h
make  all-am
make[6]: Entering directory '/home/user/nolf/openwrt/one/openwrt/build_dir/target-aarch64_cortex-a53_musl/parted-3.6/parted'
aarch64-openwrt-linux-musl-gcc  -I. -I../lib  -I../lib -I../include -I../include  -I/home/user/nolf/openwrt/one/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-15.1.0_musl/usr/include -I/home/user/nolf/openwrt/one/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-15.1.0_musl/include -I/home/user/nolf/openwrt/one/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-15.1.0_musl/include/fortify    -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/nolf/openwrt/one/openwrt/build_dir/target-aarch64_cortex-a53_musl/parted-3.6=parted-3.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -c -o command.o command.c
aarch64-openwrt-linux-musl-gcc  -I. -I../lib  -I../lib -I../include -I../include  -I/home/user/nolf/openwrt/one/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-15.1.0_musl/usr/include -I/home/user/nolf/openwrt/one/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-15.1.0_musl/include -I/home/user/nolf/openwrt/one/openwrt/staging_dir/toolchain-aar [... cut ...]
make[4]: Leaving directory '/home/user/nolf/openwrt/one/openwrt/build_dir/target-aarch64_cortex-a53_musl/parted-3.6'
make[3]: *** [Makefile:1746: all] Error 2
```

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

XZiar, 2025-07-30:
looks like a patch will be needed:
https://github.com/bcl/parted/commit/16343bda6ce0d41edf43f8dac368db3bbb63d271

pwned-pixel, 2025-07-31:
Ty, I saw this patch already. But forgot to attach. I assume that this patch needs to be integrated upstream first. But it is quite old and never got attention upstream.

Should I file this patch as a patch for parted in this repo?

Anyway I was lazy and just added `EXTRA_CFLAGS += -Wno-error=incompatible-pointer-types` to the Makefile to supress this error. But I admit that this is not wise to do because the nature of the application. 🗡️

pwned-pixel, 2025-08-11:
I've just opened an issue within their git repo asking for a new release. Let's see what happens.

https://github.com/bcl/parted/issues/17

pwned-pixel, 2025-08-12:
> Likely soon :)

is the answer. So just wait or file the patch for this repo?

neheb, 2025-08-14:
> Ty, I saw this patch already. But forgot to attach. I assume that this patch needs to be integrated upstream first. But it is quite old and never got attention upstream.
>
> Should I file this patch as [...]

doesn't matter. compiles to the same code before and after the fix.