openwrt/packages #27131: rpcsvc-proto: gettext version check error

ZqinKing, 2025-07-31:
### Package Name

rpcsvc-proto

### Maintainer

@mcprat

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

qualcommax/ipq807x

### Steps to Reproduce

Since gettext was updated to 0.24.1, in addition to its own continuous error reports, the rpcsvc-proto fails version validation during compilation.

### Actual Behaviour

make[6]: Entering directory '/home/zqinking/wrt_release/imm-nss/build_dir/hostpkg/rpcsvc-proto-1.4.3/po'
sed -e '/^#/d' remove-potcdate.sin > t-remove-potcdate.sed
mv t-remove-potcdate.sed remove-potcdate.sed
package_gnu=""; \
test -n "$package_gnu" || { \
  if { if (LC_ALL=C find --version) 2>/dev/null | grep GNU >/dev/null; then \
         LC_ALL=C find -L .. -maxdepth 1 -type f -size -10000000c -exec grep -i 'GNU rpcsvc-proto' /dev/null '{}' ';' 2>/dev/null; \
       else \
         LC_ALL=C grep -i 'GNU rpcsvc-proto' ../* 2>/dev/null; \
       fi; \
     } | grep -v 'libtool:' >/dev/null; then \
     package_gnu=yes; \
   else \
     package_gnu=no; \
   fi; \
}; \
if test "$package_gnu" = "yes"; then \
  package_prefix='GNU '; \
else \
  package_prefix=''; \
fi; \
if test -n '' || test 'kukuk@thkukuk.de' = '@'PACKAGE_BUGREPORT'@'; then \
  msgid_bugs_address=''; \
else \
  msgid_bugs_address='kukuk@thkukuk.de'; \
fi; \
case `/home/zqinking/wrt_release/imm-nss/staging_dir/hostpkg/bin/xgettext --version | sed 1q | sed -e 's,^[^0-9]*,,'` in \
  '' | 0.[0-9] | 0.[0-9].* | 0.1[0-5] | 0.1[0-5].* | 0.16 | 0.16.[0-1]*) \
    /home/zqinking/wrt_release/imm-nss/staging_dir/hostpkg/bin/xgettext --default-domain=rpcsvc-proto --directory=.. \
      --add-comments=TRANSLATORS: \
      --files-from=./POTFILES.in \
      --copyright-holder='Free Software Foundation, Inc.' \
      --msgid-bugs-address="$msgid_bugs_address" \
      --keyword=_ --keyword=N_  \
    ;; \
  *) \
    /home/zqinking/wrt_release/imm-nss/staging_dir/hostpkg/bin/xgettext --default-domain=rpcsvc-proto --directory=.. \
      --add-comments=TRANSLATORS: \
      --files-from=./POTFILES.in \
      --copyright-holder='Free Software Foundation, Inc.' \
      --package-name="${package_prefix}rpcsvc-proto" \
      --package-version='1.4.3' \
      --msgid-bugs-address="$msgid_bugs_address" \
      --keyword=_ --keyword=N_  \
    ;; \
esac
test ! -f rpcsvc-proto.po || { \
  if test -f ./rpcsvc-proto.pot-header; then \
    sed -e '1,/^#$/d' < rpcsvc-proto.po > rpcsvc-proto.1po && \
    cat ./rpcsvc-proto.pot-header rpcsvc-proto.1po > rpcsvc-proto.po && \
    rm -f rpcsvc-proto.1po \
    || exit 1; \
  fi; \
  if test -f ./rpcsvc-proto.pot; then \
    sed -f remove-potcdate.sed < ./rpcsvc-proto.pot > rpcsvc-proto.1po && \
    sed -f remove-potcdate.sed < rpcsvc-proto.po > rpcsvc-proto.2po && \
    if cmp rpcsvc-proto.1po rpcsvc-proto.2po >/dev/null 2>&1; then \
      rm -f rpcsvc-proto.1po rpcsvc-proto.2po rpcsvc-proto.po; \
    else \
      rm -f rpcsvc-proto.1po rpcsvc-proto.2po ./rpcsvc-proto.pot && \
      mv rpcsvc-proto.po ./rpcsvc-proto.pot; \
    fi; \
  else \
    mv rpcsvc-proto.po ./rpcsvc-proto.pot; \
  fi; \
}
make[6]: Leaving directory '/home/zqinking/wrt_release/imm-nss/build_dir/hostpkg/rpcsvc-proto-1.4.3/po'
*** error: gettext infrastructure mismatch: using a Makefile.in.in from gettext version 0.20 but the autoconf macros are from gettext version 0.22

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ZqinKing, 2025-07-31:
The version of gettext on the host machine is 0.21.
<pre>
(base) zqinking@zqinking:~/wrt_release/imm-nss$ command -v gettext
/usr/bin/gettext
(base) zqinking@zqinking:~/wrt_release/imm-nss$ /usr/bin/gettext -V
gettext (GNU gettext-runtime) 0.21
版权所有 (C) 1995-2020 Free Software Foundation, Inc.
授权协议 GPLv3+: GNU GPL 版本 3 或之后版本 <https://gnu.org/licenses/gpl.html>
这是自由软件: 您有修改和再发布的自由。
本软件在法律允许的范围内不提供任何担保。
由 Ulrich Drepper 编写。
</pre>

ZqinKing, 2025-07-31:
The version defined by GETTEXT_MACRO_VERSION in build_dir/hostpkg/rpcsvc-proto-1.4.3/Makefile is 0.22, which may have been generated by aclocal, but I'm not sure why it is 0.22. In build_dir/hostpkg/rpcsvc-proto-1.4.3/po/Makefile.in.in, GETTEXT_MACRO_VERSION is defined by default as 0.20.

ZqinKing, 2025-07-31:
I roughly figured it out, there exists AC_SUBST([GETTEXT_MACRO_VERSION], [0.22]) in staging_dir/host/share/aclocal/po.m4, which has not been updated due to the gettext update.

ZqinKing, 2025-07-31:
[config.log](https://github.com/user-attachments/files/21524049/config.log)

[aclocal.m4.txt](https://github.com/user-attachments/files/21524064/aclocal.m4.txt)

thkukuk, 2025-08-01:
rpcsvc-1.4.4.tar.xz package ships gettext.m4 and po/Makefile.in.in from gettext-0.20.2 and this works fine for me, even if gettext 0.22.5 is installed in the system.
This sounds like somebody is updating gettext.m4 in the source code but not po/Makefile.in.in, which means a bug in your build system.
Either fix your build system or use autogen.sh from the sources to regenerate everything.

laipeng668, 2025-08-01:
After updating gettext-full to 0.24.1, [this](https://github.com/openwrt/openwrt/commit/da541f7acd62bc33e7b0a891cf65a39d4bfe0b96) is exactly where the build system errors out for the first time. After fixing the [gettext-full/Makefile](https://github.com/openwrt/openwrt/pull/19614), here is where it fails again the second time. Seems the issue has to be traced back to the root cause after all.