openwrt/packages #27145: mwan3 PKG_VERSION:=2.12.0 bugs with luci-app-mwan3

openips, 2025-08-02:
### Package Name

mwan3 PKG_VERSION:=2.12.0 

### Maintainer

Florian Eckert <fe@dev.tdt.de>

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

mediatek/filogic

### Steps to Reproduce

1. install mwan3 PKG_VERSION:=2.12.0  and luci-app-mwan3 
2. add ipv6 interfaces  members and rules
3. show error in http://router/cgi-bin/luci/admin/status/mwan3
 

[image]

### Actual Behaviour

i find tihs bug is caused by mwan3 PKG_VERSION:=2.12.0 and same config with ipv6 work very well  in PKG_VERSION:=2.11.19
https://github.com/openwrt/luci/issues/7885

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

champtar, 2025-08-02:
@openips can you share the output of `ubus call mwan3 status '{"section":"interfaces"}'` with mwan3 `2.11.19` and `2.12.0`

openips, 2025-08-02:
@champtar  please check and find any error by  the output of ubus call mwan3 status '{"section":"interfaces"}' with mwan3 2.11.19 and 2.12.0

champtar, 2025-08-02:
If you can share network and mwan3 config file that would be great, else I'll setup a VM to reproduce (I don't use IPv6 with my mwan3 routers)

openips, 2025-08-02:
[mwan3_config.zip](https://github.com/user-attachments/files/21562804/mwan3_config.zip)
I upload my   network and mwan3 config file , marked my username and password as ****** in network config file.
Please replace your account and password to see if you can find the error and fix the bug. Thank you.

champtar, 2025-08-02:
https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#interface_configuration

> Interface name 	string 	yes 	(none) 	The OpenWrt interface name as defined in /etc/config/network. Do not use interface names like pppoe-wanX

In your mwan3 config you have for example `cu_wan_6` interface, but it doesn't exists in network config, I need to fix this case that used to work.

openips, 2025-08-03:
> https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#interface_configuration
>
>
> In your mwan3 config you have for example `cu_wan_6` interface, but it doesn't exists in network config, I [...]

The ipv6 network interface is a virtual interface that is automatically created after the PPPoE authentication is completed.

[image]

champtar, 2025-08-03:
Ok I don't have a pppoe wan, so I can't replicate easily
Can you share the output of
- `ubus call network.interface dump | grep cu_wan`
- `ubus list | grep cu_wan`

Also do you have any interesting logs (rpcd) in logread

openips, 2025-08-03:
```
root@BPI_R4:~# ubus call network.interface dump | grep cu_wan
                        "interface": "cu_wan",
                        "l3_device": "pppoe-cu_wan",
                        "interface": "cu_wan_6",
                        "l3_device": "pppoe-cu_wan",
                        "device": "pppoe-cu_wan",
                                        "class": "cu_wan_6",
```
```
root@BPI_R4:~# ubus list | grep cu_wan
network.interface.cu_wan
network.interface.cu_wan_6
```

openips, 2025-08-03:
> Ok I don't have a pppoe wan, so I can't replicate easily Can you share the output of
> 
> * `ubus call network.interface dump | grep cu_wan`
> * `ubus list | grep cu_wan`
> 
> Also do you have any interesting logs (rpcd) in logread

I didn't find any special errors in the logread

champtar, 2025-08-03:
@openips please try #27147

openips, 2025-08-03:
ok i will try later

openips, 2025-08-03:
> [@openips](https://github.com/openips) please try [#27147](https://github.com/openwrt/packages/pull/27147)

After using the patch, the error still appears

goto http://router/cgi-bin/luci/admin/status/mwan3
show error
```
RPCError
RPC call to mwan3/status failed with error -32000: Object not found
  at ClassConstructor.handleCallReply (http://192.168.1.1/luci-static/resources/rpc.js?v=25.213.48820~3ec9252-1754084853:11:3)
```

[image]

champtar, 2025-08-03:
I now have a x86_64 OpenWRT master VM with just luci-ssl and luci-app-mwan3 and I can't reproduce your error :(
(I don't have pppoe but just creating a non existent interface)

Can you run `ucode -p 'mwan3 = loadfile("/usr/share/rpcd/ucode/mwan3"); print(mwan3().mwan3.status.call({"args":{"section":"interfaces"}}))'`

openips, 2025-08-04:
OK
I find your pr is merged into last master tree. I will build a new img and then test again. Thanks

openips, 2025-08-04:
> I now have a x86_64 OpenWRT master VM with just luci-ssl and luci-app-mwan3 and I can't reproduce your error :( (I don't have pppoe but just creating a non existent interface)
>
> Can you run `ucode -p [...]

I flash a new img with your pr, mwan3 still can not show ipv6 interface

```
root@BPI_R4:~# ucode -p 'mwan3 = loadfile("/usr/share/rpcd/ucode/mwan3"); print(mwan3().mwan3.status.call({"args":{"section":"interfaces"}}))'
Reference error: left-hand side expression is null
In [arrow function](), file /usr/share/rpcd/ucode/mwan3, line 144, byte 24:
  called from function foreach ([C])
  called from function interfaces_status (/usr/share/rpcd/ucode/mwan3:155:3)
  called from function [anonymous function] (/usr/share/rpcd/ucode/mwan3:183:46)
  called from anonymous function ([-p argument]:1:115)

 `            'uptime': netstatus.uptime || 0,`
  Near here ----------------------^

```

I wonder if you can add an ipv6 network interface in your virtual machine to reproduce this problem. Thank you for your work. @champtar

champtar, 2025-08-04:
@openips you are not running latest mwan3 version with my PR, `netstatus.uptime` doesn't exists anymore, it's
https://github.com/openwrt/packages/blob/ac1076ef9530138e7c84d2858d5cf2b5d8c74e02/net/mwan3/files/usr/share/rpcd/ucode/mwan3#L144

openips, 2025-08-05:
Your pr is successfully merged into master tree .So I use  master tree to make lastest mwan3 img.

I use mwan3 below
[mwan3.txt](https://github.com/user-attachments/files/21586326/mwan3.txt)
from https://github.com/openwrt/packages/blob/master/net/mwan3/files/usr/share/rpcd/ucode/mwan3
in master package tree,mwan3 L144 is
https://github.com/openwrt/packages/blob/ac1076ef9530138e7c84d2858d5cf2b5d8c74e02/net/mwan3/files/usr/share/rpcd/ucode/mwan3#L144
is your pr ,mwan3 L144 is
https://github.com/champtar/packages/blob/f891179288e5ffb6650588db30f9c44e9a96979d/net/mwan3/files/usr/share/rpcd/ucode/mwan3#L144

they are same.

closed 2025-08-05 by openips