openwrt/packages #27146: adblock-fast: `Unknown text` error and warnings shown during startup

qosmio, 2025-08-02:
### Package Name

adblock-fast

### Maintainer

@stangri

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

qualcommax/ipq807x

### Steps to Reproduce

Issue related to commit: a94c5fac4b738bb52f0d627842744152f54108d5

Using the following config (left out all `option enabled '0'` feeds)
```vim
config adblock-fast 'config'
	option enabled '1'
	option allow_non_ascii '0'
	option canary_domains_icloud '0'
	option canary_domains_mozilla '0'
	option compressed_cache '1'
	option compressed_cache_dir '/var/lib/adblock'
	option config_update_enabled '1'
	option config_update_url 'https://cdn.jsdelivr.net/gh/openwrt/packages/net/adblock-fast/files/adblock-fast.config.update'
	option curl_max_file_size '30000000'
	option curl_retry '3'
	option debug '0'
	option dns 'unbound.adb_list'
	list dnsmasq_instance '*'
	option download_timeout '10'
	option force_dns '0'
	list force_dns_port '53'
	list force_dns_port '853'
	option parallel_downloads '1'
	option pause_timeout '20'
	option procd_trigger_wan6 '0'
	option procd_boot_delay '0'
	option procd_boot_wan_timeout '60'
	option verbosity '1'
	option heartbeat_domain 'heartbeat.melmac.ca'
	option heartbeat_sleep_timeout '10'
	option sanity_check '1'
	option update_config_sizes '1'
	list allowed_domain 'cdn.jsdelivr.net'
	list allowed_domain 'emjcd.com'
	list allowed_domain 'prf.hn'
	list allowed_domain 'bestbuy.7tiv.net'
	list allowed_domain 'redirectingat.com'
	list allowed_domain 'linksynergy.com'
	list allowed_domain 'jdoqocy.com'
	list allowed_domain 'dpbolvw.net'

config file_url
	option url 'https://cdn.jsdelivr.net/gh/StevenBlack/hosts/hosts'
	option size '6551341'
	option action 'block'

config file_url
	option url 'https://raw.githubusercontent.com/AdguardTeam/cname-trackers/master/data/combined_disguised_trackers_justdomains.txt'
	option size '2498459'
	option action 'block'

config file_url
	option url 'https://cdn.jsdelivr.net/gh/kboghdady/youTube_ads_4_pi-hole/black.list'
	option size '553006'
	option action 'block'
```

### Actual Behaviour

During boot up seeing the following in the logs

```node
Sat Aug  2 16:17:48 2025 user.notice adblock-fast [7208]: [TRIG] Setting trigger (on_boot) [✓]
Sat Aug  2 16:18:04 2025 user.notice adblock-fast [11817]: [INIT] Starting adblock-fast 1.1.4-r4...
Sat Aug  2 16:18:04 2025 user.notice adblock-fast [11817]: [STAT] adblock-fast 1.1.4-r4 is blocking 189579 domains (with unbound.adb_list)
Sat Aug  2 16:18:04 2025 user.notice adblock-fast [11817]: [ERROR] Unknown text ''!
Sat Aug  2 16:18:04 2025 user.notice adblock-fast [11817]: [ERROR] Unknown text ''!
Sat Aug  2 16:18:04 2025 user.notice adblock-fast [11817]: [ERROR] Unknown text ''!
Sat Aug  2 16:18:04 2025 user.notice adblock-fast [11817]: [ERROR] Unknown text ''!
Sat Aug  2 16:18:04 2025 user.notice adblock-fast [11817]: [ERROR] Unknown text ''!
Sat Aug  2 16:18:04 2025 user.notice adblock-fast [11817]: [WARN] Unknown text ''!
Sat Aug  2 16:18:04 2025 user.notice adblock-fast [11817]: [WARN] Unknown text ''!
Sat Aug  2 16:18:04 2025 user.notice adblock-fast [11817]: [WARN] Unknown text ''!
Sat Aug  2 16:18:05 2025 user.notice adblock-fast [11817]: [WARN] Unknown text ''!
Sat Aug  2 16:18:05 2025 user.notice adblock-fast [11817]: [WARN] Unknown text ''!
Sat Aug  2 16:18:05 2025 user.notice adblock-fast [11817]: [TRIG] Setting trigger for wan [✓]
Sat Aug  2 16:18:14 2025 user.notice adblock-fast [13596]: [INIT] Force-reloading adblock-fast 1.1.4-r4...
Sat Aug  2 16:18:17 2025 user.notice adblock-fast [13596]: [ DL ] Blocked List: cdn.jsdelivr.net (domains) [✓]
Sat Aug  2 16:18:17 2025 user.notice adblock-fast [13596]: [ DL ] Blocked List: raw.githubusercontent.com (domains) [✓]
Sat Aug  2 16:18:19 2025 user.notice adblock-fast [13596]: [ DL ] Blocked List: cdn.jsdelivr.net (hosts) [✓]
Sat Aug  2 16:18:20 2025 user.notice adblock-fast [13596]: [PROC] Sorting combined block-list [✓]
Sat Aug  2 16:18:20 2025 user.notice adblock-fast [13596]: [PROC-DEBUG] Sorting combined block-list took 1s
Sat Aug  2 16:19:24 2025 user.notice adblock-fast [13596]: [PROC] Optimizing combined block-list [✓]
Sat Aug  2 16:19:24 2025 user.notice adblock-fast [13596]: [PROC-DEBUG] Optimizing combined block-list took 64s
Sat Aug  2 16:19:25 2025 user.notice adblock-fast [13596]: [PROC] Removing allowed domains from combined block-list [✓]
Sat Aug  2 16:19:25 2025 user.notice adblock-fast [13596]: [PROC-DEBUG] Removing allowed domains from combined block-list took 1s
Sat Aug  2 16:19:25 2025 user.notice adblock-fast [13596]: [PROC] Formatting combined block-list file [✓]
Sat Aug  2 16:19:25 2025 user.notice adblock-fast [13596]: [PROC-DEBUG] Formatting combined block-list file took 0s
Sat Aug  2 16:19:25 2025 user.notice adblock-fast [13596]: [PROC] Setting up unbound.adb_list file [✓]
Sat Aug  2 16:19:26 2025 user.notice adblock-fast [13596]: [PROC] Creating unbound.adb_list compressed cache [✓]
Sat Aug  2 16:19:26 2025 user.notice adblock-fast [13596]: [PROC-DEBUG] Creating unbound.adb_list compressed cache took 0s
Sat Aug  2 16:19:26 2025 user.notice adblock-fast [13596]: [PROC] Removing temporary files [✓]
Sat Aug  2 16:19:26 2025 user.notice adblock-fast [13596]: [UNBD] Updating unbound configuration [✓]
Sat Aug  2 16:19:26 2025 user.notice adblock-fast [13596]: [UNBD] Sanity check for unbound.adb_list TLDs [✓]
Sat Aug  2 16:19:26 2025 user.notice adblock-fast [13596]: [UNBD] Sanity check for unbound.adb_list leading dots [✓]
Sat Aug  2 16:19:28 2025 user.notice adblock-fast [13596]: [UNBD] Restarting unbound [✓]
Sat Aug  2 16:19:30 2025 user.notice adblock-fast [13596]: [UNBD] Probing heartbeat.melmac.ca for 10 seconds [w][✓]
Sat Aug  2 16:19:31 2025 user.notice adblock-fast [13596]: [STAT] adblock-fast 1.1.4-r4 is blocking 189579 domains (with unbound.adb_list)
Sat Aug  2 16:19:31 2025 user.notice adblock-fast [13596]: [TRIG] Setting trigger for wan [✓]
```

The same messages are not seen after a restart or a reload:
```node
Sat Aug  2 15:54:33 2025 user.notice adblock-fast [20022]: Force-reloading adblock-fast 1.1.4-r4...
Sat Aug  2 15:54:36 2025 user.notice adblock-fast [20022]: Downloading lists ✓✓✓
Sat Aug  2 15:54:37 2025 user.notice adblock-fast [20022]: [PROC-DEBUG] Sorting combined block-list took 1s
Sat Aug  2 15:54:41 2025 user.notice adblock-fast [20022]: [PROC-DEBUG] Optimizing combined block-list took 4s
Sat Aug  2 15:54:41 2025 user.notice adblock-fast [20022]: [PROC-DEBUG] Removing allowed domains from combined block-list took 0s
Sat Aug  2 15:54:42 2025 user.notice adblock-fast [20022]: [PROC-DEBUG] Formatting combined block-list file took 1s
Sat Aug  2 15:54:46 2025 user.notice adblock-fast [20022]: [PROC-DEBUG] Creating unbound.adb_list compressed cache took 0s
Sat Aug  2 15:54:46 2025 user.notice adblock-fast [20022]: Processing downloads ✓✓✓✓✓✓✓
Sat Aug  2 15:54:50 2025 user.notice adblock-fast [20022]: Cycling unbound ✓✓✓✓.✓
Sat Aug  2 15:54:51 2025 user.notice adblock-fast [20022]: * adblock-fast 1.1.4-r4 is blocking 189579 domains (with unbound.adb_list)
Sat Aug  2 15:54:51 2025 user.notice adblock-fast [20022]: Setting trigger for wan ✓
```

It doesn't seem to affect the service though.

As a very crude debug method, I added
```diff
@@ -2454,6 +2454,7 @@ adb_status() {
        [ "$param" != 'quiet' ] || return 0
        if [ -n "$error" ]; then
                for c in $error; do
+                 echo error "$c" >> /tmp/err
                        code="$(json get error "$c" 'code')"
                        info="$(json get error "$c" 'info')"
                        output_error "$(get_text "$code" "$info")"
@@ -2461,6 +2462,7 @@ adb_status() {
        fi
        if [ -n "$warning" ]; then
                for c in $warning; do
+                 echo warn "$c" >> /tmp/err
                        code="$(json get warning "$c" 'code')"
                        info="$(json get warning "$c" 'info')"
                        output_warning "$(get_text "$code" "$info")"
```

```
error status
error version
error packageCompat
error message
error stats
warn status
warn version
warn packageCompat
warn message
warn stats
```

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

stangri, 2025-08-03:
Thanks for digging into it, should be fixed by changing line 462 in json():
```sh
						json_select "$param" >/dev/null 2>&1
```
-->
```sh
						json_select "$param" >/dev/null 2>&1 || return
```

Sorry, that was addressed during the PR for pbr, but not adblock-fast.

stangri, 2025-08-05:
Thanks for a quick test, I have no idea why it didn't come up in pre-PR testing, but I'll create another PR with a fix.

closed 2025-08-05 by stangri