openwrt/packages #27213: owut: ucv_resource_new_ex: symbol not found (looks in file /usr/lib/ucode/uloop.so

kvaml, 2025-08-11:
### Package Name

owut

### Maintainer

Eric Fahlgren <ericfahlgren@gmail.com>

### OpenWrt Version

24.10.0

### OpenWrt Target/Subtarget

mvebu/cortexa53

### Steps to Reproduce

1. opkg install owut
2. owut does not work.  It is unable to even report the version
3. 
# opkg info owut
Package: owut
Version: 2025.07.11~0d00192d-r1
Depends: libc, attendedsysupgrade-common, rpcd-mod-file, ucode, ucode-mod-fs, ucode-mod-ubus, ucode-mod-uci, ucode-mod-uclient, ucode-mod-uloop
Status: install user installed
Section: utils
Architecture: all
Size: 28183
Filename: owut_2025.07.11~0d00192d-r1_all.ipk
Conffiles:
 /etc/owut.d/pre-install.sh 91770a2f553d58c0697d9dbf22244f944a6963f414e48dc1d0e8384644e8ec1a
Description: The OpenWrt Upgrade Tool (owut) is a command line program that
 checks for new packages, package build breakages, generates
 sysupgrade images containing installed packages and installs
 images.  It is written in ucode for easy end-user customization.
 
 Documentation is available at
 https://openwrt.org/docs/guide-user/installation/sysupgrade.owut
Installed-Time: 1754873948

Since it appears to relate to ucode-mod-uloop

# opkg info ucode-mod-uloop
Package: ucode-mod-uloop
Version: 2025.07.18~3f64c808-r1
Depends: libc, ucode, libubox20240329
Status: install ok installed
Section: lang
Architecture: aarch64_cortex-a53
Size: 10259
Filename: ucode-mod-uloop_2025.07.18~3f64c808-r1_aarch64_cortex-a53.ipk
Description: The uloop module allows ucode scripts to interact with OpenWrt uloop event loop implementation.
Installed-Time: 1754873948

### Actual Behaviour

root@OpenWrt:/# owut -V
Runtime error: Unable to dlopen file '/usr/lib/ucode/uloop.so': Error relocating /usr/lib/ucode/uloop.so: ucv_resource_new_ex: symbol not found
In /usr/bin/owut, line 10, byte 34:

 `let PROG    = `${NAME}/${VERSION}`;`
  Near here -----------------------^

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

efahl, 2025-08-13:
That's a packaging error with `ucode` and its modules:
https://forum.openwrt.org/t/owut-openwrt-upgrade-tool/200035/691

My current suggestion is to use LuCI Attended Sysupgrade to do this one upgrade, as it doesn't use `ucode-mod-uloop` and thus should work past this error.  (If you don't have space on-device for `luci-app-attendedsysupgrade` or don't want to install it, you can use https://firmware-selector.openwrt.org, but it's a lot more work.)

kvaml, 2025-08-14:
Thanks.  The LuCI AttendedSysupgrade never completed.  That's what pushed me to owut.  Thanks for suggesting that work-around.  I will give it a shot.

lijunle, 2025-10-21:
> ```
> opkg upgrade ucode libucode
> ```

I am getting bad gateway in Luci. This saved the world!

> Bad Gateway
> The process did not produce any response