openwrt/packages #27228: <package-name>: <short description>

EndProgramm, 2025-08-12:
### Package Name

pgsql-server

### Maintainer

@dangowrt

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

mediatek/filogic

### Steps to Reproduce

Install the package

or

Run ´pg_ctl init -U postgres -D <folder for database cluster>´

### Actual Behaviour

see [this forum post](https://forum.openwrt.org/t/postgresql-server-package-not-able-to-create-database-cluster/238923?u=endos)

TLDR:
the setup fails for any folder with output like this:
```bash
root@BananeN:~# sudo -u postgres pg_ctl init -U postgres -D /var/postgres/data
The files belonging to this database system will be owned by user "postgres".
This user must also own the server process.

The database cluster will be initialized with this locale configuration:
  locale provider:   libc
  LC_COLLATE:  C
  LC_CTYPE:    C.UTF-8
  LC_MESSAGES: C
  LC_MONETARY: C
  LC_NUMERIC:  C
  LC_TIME:     C
The default database encoding has accordingly been set to "UTF8".
The default text search configuration will be set to "english".

Data page checksums are disabled.

fixing permissions on existing directory /var/postgres/data ... ok
creating subdirectories ... ok
selecting dynamic shared memory implementation ... posix
selecting default "max_connections" ... 100
selecting default "shared_buffers" ... 128MB
selecting default time zone ... CET
creating configuration files ... ok
running bootstrap script ... ok
performing post-bootstrap initialization ... sh: locale: not found
2025-08-11 19:29:38.736 CEST [15225] WARNING:  no usable system locales were found
2025-08-11 19:29:38.741 CEST [15225] FATAL:  could not load library "/usr/lib/postgresql/dict_snowball.so": Error relocating /usr/lib/postgresql/dict_snowball.so: pg_any_to_server: symbol not found
2025-08-11 19:29:38.741 CEST [15225] STATEMENT:  CREATE FUNCTION dsnowball_init(INTERNAL)
            RETURNS INTERNAL AS '$libdir/dict_snowball', 'dsnowball_init'
        LANGUAGE C STRICT;

child process exited with exit code 1
initdb: removing contents of data directory "/var/postgres/data"
pg_ctl: database system initialization failed
```

When insepecting the file `/usr/lib/postgresql/dict_snowball.so` with `ldd`, it outputs this:
```bash
root@BananeN:~# ldd /usr/lib/postgresql/dict_snowball.so 
        ldd (0x7f8ec40000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f8eb2e000)
        libc.so => ldd (0x7f8ec40000)
Error relocating /usr/lib/postgresql/dict_snowball.so: pg_any_to_server: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: lowerstr: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: pfree: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: errmsg: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: errfinish: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: palloc0: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: pg_server_to_any: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: GetDatabaseEncoding: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: CurrentMemoryContext: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: lowerstr_with_len: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: errstart_cold: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: errcode: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: GetDatabaseEncodingName: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: palloc: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: readstoplist: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: pg_strcasecmp: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: defGetString: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: searchstoplist: symbol not found
Error relocating /usr/lib/postgresql/dict_snowball.so: repalloc: symbol not found
```

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Yang-Wei-Ting, 2025-12-15:
@ppvan  
Does the above PR fix the bug you encountered?

danielfdickinson, 2025-12-15:
@ppvan For `/opt/lib/postgresql/plpgsql.so`, `/opt/lib` is not a standard OpenWrt library path. Can you elaborate on how you are installing postgresql and any related (postgresql, opkg, or apk) configuration? Also the version of OpenWrt for which you are experiencing this issue? (And in that vein is it 'stock' OpenWrt or a vendor-modified version?)