openwrt/packages #27330: net/rsync: rrsync pulls in Perl, but it's a Python program

mologie, 2025-08-24:
### Package Name

rsync

### Maintainer

@mstorchak

### OpenWrt Version

24.10 and latest

### OpenWrt Target/Subtarget

all

### Steps to Reproduce

```
$ opkg install rrsync && rrsync
Installing rrsync (3.4.1-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/rrsync_3.4.1-r1_aarch64_generic.ipk
Installing perl (5.40.0-r2) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/perl_5.40.0-r2_aarch64_generic.ipk
Installing perlbase-config (5.40.0-r2) to root...
[...]
env: ‘python3’: No such file or directory
```

### Actual Behaviour

The tool does not run because its package depends on Perl, but `/usr/bin/rrsync` is a Python script. Python is not installed by default. Oops, happens, it used to be Perl a long time ago after all.

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

mstorchak, 2025-08-26:
Thanks for the notice @mologie 
Since you seem to be interested in the script, could you please find the correct set of python packages to run it? Is python3 enough?

mologie, 2025-08-30:
Thanks for fixing this, really appreciate it