openwrt/packages #27331: rust: old builds get deleted after certain time /  failed to download llvm from ci

brocaar, 2025-08-25:
### Package Name

rust

### Maintainer

@lu-zero

### OpenWrt Version

24.10.2

### OpenWrt Target/Subtarget

bcm27xx/bcm2712

### Steps to Reproduce

1. Checkout OpenWrt version v24.10.2
2. Make sure there is no download / build cache
3. Use `src-git packages https://git.openwrt.org/feed/packages.git^c7d1a8c1ae976bd0ad94a351d82ee8fbf16a81f0` in the feed (this is the same commit as that v24.10.2 points to)
4. Try to build any package that uses `rust`

### Actual Behaviour

Building `rust` 1.85.0 fails with the following error:

```
make[3]: Entering directory '/workdir/openwrt/feeds/packages/lang/rust'
CARGO_HOME=/workdir/openwrt/dl/cargo TARGET_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a76_musl/rust-1.85.0=rust-1.85.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" python3 /workdir/openwrt/build_dir/target-aarch64_cortex-a76_musl/host/rustc-1.85.0-src/x.py --build-dir /workdir/openwrt/build_dir/target-aarch64_cortex-a76_musl/host/rustc-1.85.0-src/build --config /workdir/openwrt/build_dir/target-aarch64_cortex-a76_musl/host/rustc-1.85.0-src/config.toml dist build-manifest cargo llvm-tools rustc rust-std rust-src
Building bootstrap
    Finished `dev` profile [unoptimized] target(s) in 0.29s
WARNING: The `change-id` is missing in the `config.toml`. This means that you will not be able to track the major changes made to the bootstrap configurations.
NOTE: to silence this warning, add `change-id = 134650` at the top of `config.toml`
downloading https://ci-artifacts.rust-lang.org/rustc-builds/4d91de4e48198da2e33413efdcd9cd2cc0c46688/rust-dev-1.85.0-x86_64-unknown-linux-gnu.tar.xz
curl: (22) The requested URL returned error: 404
Warning: Problem (retrying all errors). Will retry in 1 seconds. 3 retries
Warning: left.
curl: (22) The requested URL returned error: 404
Warning: Problem (retrying all errors). Will retry in 2 seconds. 2 retries
Warning: left.
curl: (22) The requested URL returned error: 404
Warning: Problem (retrying all errors). Will retry in 4 seconds. 1 retries
Warning: left.
curl: (22) The requested URL returned error: 404

ERROR: failed to download llvm from ci

    HELP: There could be two reasons behind this:
        1) The host triple is not supported for `download-ci-llvm`.
        2) Old builds get deleted after a certain time.
    HELP: In either case, disable `download-ci-llvm` in your config.toml:

    [llvm]
    download-ci-llvm = false

Build completed unsuccessfully in 0:00:08
make[3]: *** [Makefile:109: /workdir/openwrt/build_dir/target-aarch64_cortex-a76_musl/host/rustc-1.85.0-src/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#0.42#0.28#8.54
    ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:174: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:170: /workdir/openwrt/staging_dir/target-aarch64_cortex-a76_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:233: world] Error 2
```

---

It looks it is only possible to use the latest-ish version of Rust. After some time, it is no longer be possible to build Rust from source since the needed files get removed from `ci-artifacts.rust-lang.org`, after which you get:

```
ERROR: failed to download llvm from ci

    HELP: There could be two reasons behind this:
        1) The host triple is not supported for `download-ci-llvm`.
        2) Old builds get deleted after a certain time.
    HELP: In either case, disable `download-ci-llvm` in your config.toml:

    [llvm]
    download-ci-llvm = false
```

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

lu-zero, 2025-08-25:
I think we agreed to not use the llvm prebuilt. 

@1715173329 do we have consensus in building a single llvm for target, host and ebpf as part of the toolchain?

evertonfraga0, 2026-01-17:
any solutions?

brocaar, 2026-03-04:
Thanks for your feedback @lu-zero :)

closed 2026-03-05 by GeorgeSapkin