openwrt/packages #27332: nft-qos: Why Porn site is owner of nft-qos .!!

lessload, 2025-08-25:
### Package Name

nft-qos

### Maintainer

rosysong@rosinson.com

### OpenWrt Version

24.10.2

### OpenWrt Target/Subtarget

ramips/mt7621

### Steps to Reproduce

Why nft-qos was own by port site ?? rosysong@rosinson.com seem like they was gone ?
As i check rosinson.com was a porn site .! no one check it .!?

### Actual Behaviour

openwrt team should have new maintainer for it and remove anything about rosinson.com (porn site) from source code.

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

lessload, 2025-08-26:
> I don't understand why rosinson.com being an adult site is a problem. 

It not just a normal porn site , it is porn site sponsor by bad gambling that relate to scammer in china, that why i feel scary when found it in openwrt source code. it feel like some code of openwrt owned by scammer.

brada4, 2025-08-29:
Squatter took over abandoned domain.
Ever heard of wayback machine before attacking owners of it from 5yrs ago?

brada4, 2025-09-03:
Seeing https://github.com/openwrt/packages/issues/16007 seems abandoned upstream.
It is partially working at best. Not broken, but most config params are no-ops.