openwrt/packages #27357: lsof: does not compile on current master / not packaged on builbot

pwned-pixel, 2025-08-29:
### Package Name

lsof

### Maintainer

Maxim Storchak <m.storchak@gmail.com>

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

x86/64

### Steps to Reproduce

Lsof got broken a while ago. The buildbot is not packaging it on SNAPSHOT atm.
I tried to compile for targets x86_64 and qualcommax/ipq807x. Both failed.

### Actual Behaviour

```
OpenWrt-libtool: link: aarch64-openwrt-linux-musl-gcc -fvisibility=hidden -O3 -pipe -mcpu=cortex-a53+crc+crypto -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/openwrt/zyxel_nbg7815/openwrt-nss-28-08-2025-2007/build_dir/target-aarch64_cortex-a53_musl/lsof-4.99.5=lsof-4.99.5 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -D_FILE_OFFSET_BITS=64 -I/home/user/openwrt/zyxel_nbg7815/openwrt-nss-28-08-2025-2007/staging_dir/target-aarch64_cortex-a53_musl/usr/include/tirpc -fuse-ld=bfd -Wl,--gc-sections -znow -zrelro -o lsof src/lsof-arg.o src/lsof-main.o src/lsof-print.o src/lsof-ptti.o src/lsof-store.o src/lsof-usage.o src/lsof-util.o src/dialects/linux/lsof-dprint.o lib/lsof-ckkv.o lib/lsof-cvfs.o lib/lsof-dvch.o lib/lsof-fino.o lib/lsof-isfn.o lib/lsof-lkud.o lib/lsof-lsof.o lib/lsof-misc.o lib/lsof-node.o lib/lsof-pdvn.o lib/lsof-prfp.o lib/lsof-print.o lib/lsof-proc.o lib/lsof-ptti.o lib/lsof-rdev.o lib/lsof-rnmt.o lib/lsof-rmnt.o lib/lsof-rnam.o lib/lsof-rnch.o lib/lsof-rnmh.o lib/dialects/linux/lsof-dfile.o lib/dialects/linux/lsof-dmnt.o lib/dialects/linux/lsof-dnode.o lib/dialects/linux/lsof-dproc.o lib/dialects/linux/lsof-dsock.o lib/dialects/linux/lsof-dstore.o  -L/home/user/openwrt/zyxel_nbg7815/openwrt-nss-28-08-2025-2007/staging_dir/toolchain-aarch64_cortex-a53_gcc-15.2.0_musl/usr/lib -L/home/user/openwrt/zyxel_nbg7815/openwrt-nss-28-08-2025-2007/staging_dir/toolchain-aarch64_cortex-a53_gcc-15.2.0_musl/lib -L/home/user/openwrt/zyxel_nbg7815/openwrt-nss-28-08-2025-2007/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -ltirpc
/home/user/openwrt/zyxel_nbg7815/openwrt-nss-28-08-2025-2007/staging_dir/toolchain-aarch64_cortex-a53_gcc-15.2.0_musl/bin/../lib/gcc/aarch64-openwrt-linux-musl/15.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld.bfd: src/lsof-print.o: in function `lkup_port':
print.c:(.text.lkup_port+0x290): undefined reference to `getrpcbynumber'
/home/user/openwrt/zyxel_nbg7815/openwrt-nss-28-08-2025-2007/staging_dir/toolchain-aarch64_cortex-a53_gcc-15.2.0_musl/bin/../lib/gcc/aarch64-openwrt-linux-musl/15.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld.bfd: print.c:(.text.lkup_port+0x294): undefined reference to `getrpcbynumber'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:1914: lsof] Error 1
make[4]: Leaving directory '/home/user/openwrt/zyxel_nbg7815/openwrt-nss-28-08-2025-2007/build_dir/target-aarch64_cortex-a53_musl/lsof-4.99.5'
make[3]: *** [Makefile:1264: all] Error 2
```
Just for completion: I'm using gcc 15/bin 2.45 and -O3. But this should not matter as buildbot is still on gcc 14 and does not come up with an apk package also.

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

pwned-pixel, 2025-08-29:
Ty, it worked! Do you want me to create a PR (if its a viable solution) or just leave this issue open?

brada4, 2025-08-29:
Absolutely, go for it. I just made a guess on parameter name based on missing symbol name ;-)

mstorchak, 2025-08-29:
@pwned-pixel , thank you for the notice and for the PR.
I don't quite understand how this issue happens. libtirpc is a package dependency, so configure should detect its presence correctly. Do you have any idea on why this doesn't happen?

Note: the issue might be related to #27334

As for the fix, I'm not convinced we need tirpc support, see https://github.com/openwrt/packages/pull/26896#issuecomment-3044860713 for the difference

mstorchak, 2025-08-29:
@pwned-pixel , libtirpc is optional for lsof, and according to my observations, it doesn't add much. I agree to drop RPC support from lsof, please see my comment to the PR.
As to me, if the updated libtirpc works for other users but breaks lsof, it's not worth investing in lsof-libtripc integration, and it's easier to drop RPC support from lsof. So, instead of revering the libtirpc commit, please test lsof with both your change and my proposal.

brada4, 2025-08-29:
tirpc is to list files opened via network export ie router being a file server.

ptpt52, 2025-08-30:
should fixed by https://github.com/openwrt/packages/commit/6736b2d19daca25ad691e99284233b55d11957ab

and we should revert https://github.com/openwrt/packages/commit/6b27cc517066eea1d81c3b04fa6e55ca6a061d42 ?

pwned-pixel, 2025-08-31:
> and we should revert [6b27cc5](https://github.com/openwrt/packages/commit/6b27cc517066eea1d81c3b04fa6e55ca6a061d42) ?

was  wondering the same, but the origin title of the was about fixing compiling issues with regarding gcc14 (IIRC). but i have the "feeling" that this lib has more to come for lsof and other packages.

i can try to see if it would compile after reverting if it is worth to do. but @brada wrote already it can be bigger. so prop. its not worth to do.

brada4, 2025-08-31:
My rationale is that much smaller fuser (psmisc package) does the simple listing ootb

pwned-pixel, 2025-09-01:
> Let's keep building lsof without libtirpc.

So I'll close this issue.

closed 2025-09-01 by pwned-pixel