openwrt/packages #27363: logd: loses "millis" in timestamps

ralfbergs, 2025-08-30:
### Package Name

logd

### Maintainer

(@commodo)

### OpenWrt Version

24.10.2

### OpenWrt Target/Subtarget

ramips/mt7621

### Steps to Reproduce

1. Set up OpenWrt to log to a remote syslog server

### Actual Behaviour

Observe that the timestamps used lack the millis, i.e. all millis are discarded, and `.000` is used instead.

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ralfbergs, 2025-08-31:
> logd is part of the ubox repository.
> 
> https://github.com/openwrt/ubox/issues

Many thanks. I'll change it accordingly.

dave14305, 2025-08-31:
Try adding the `-t` parameter to the logread command used to send to remote IPs.

dave14305, 2025-09-02:
Ok, you should still open a new issue in the main repo since ubox/logd/logread are not community packages from this packages repo.