openwrt/packages #27374: schroot: build fails missing boost-system package

ptpt52, 2025-09-01:
### Package Name

schroot

### Maintainer

@javier

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

all

### Steps to Reproduce

select schroot package and build

### Actual Behaviour

log:
```
touch /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot/.prepared_d9a240aef24c9e164cd19159093522db_6664517399ebbbc92a37c5bb081b5c53
rm -f /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot/.configured_*
rm -f /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl/stamp/.schroot_installed
mkdir -p /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot/build
(cd /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot/build; CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot=reschroot -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot=reschroot -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " LDFLAGS="-L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/usr/lib -L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/lib -fuse-ld=bfd -znow -zrelro   " cmake --no-warn-unused-cli -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=x86_64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/usr/lib -L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/lib -fuse-ld=bfd -znow -zrelro  " -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/usr/lib -L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/lib -fuse-ld=bfd -znow -zrelro   -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/usr/lib -L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/lib -fuse-ld=bfd -znow -zrelro   -Wl,-Bsymbolic-functions" -DCMAKE_AR="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl/usr;/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl -DCMAKE_PREFIX_PATH=/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE -DCMAKE_GENERATOR="Ninja" -Dbtrfs-snapshot=ON -Dloopback=ON -Dlvm-snapshot=ON -Duuid=ON -Dblock-device=ON -Dunion=ON -Ddhcroot=OFF -Ddchroot-dsa=OFF -Ddebug=OFF -Ddoxygen=OFF -Dpam=OFF -Dtest=OFF -Duuid=OFF -Dzfs-snapshot=OFF /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot )
Not searching for unused variables given on the command line.
CMake Deprecation Warning at CMakeLists.txt:16 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

CMake Deprecation Warning at CMakeLists.txt:17 (cmake_policy):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- The C compiler identification is GNU 14.3.0
-- The CXX compiler identification is GNU 14.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Reading release metadata from cmake/GitVersion.cmake
-- Configuring schroot 1.6.13 (15 Aug 2022)
-- Performing Test CXX_FLAG_CXX11
-- Performing Test CXX_FLAG_CXX11 - Success
-- Performing Test CXX_FLAG-pedantic
-- Performing Test CXX_FLAG-pedantic - Success
-- Performing Test CXX_FLAG-Wall
-- Performing Test CXX_FLAG-Wall - Success
-- Performing Test CXX_FLAG-Wcast-align
-- Performing Test CXX_FLAG-Wcast-align - Success
-- Performing Test CXX_FLAG-Wwrite-strings
-- Performing Test CXX_FLAG-Wwrite-strings - Success
[... 53 lines cut ...]

Call Stack (most recent call first):
  /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl/usr/lib/cmake/Boost-1.89.0/BoostConfig.cmake:262 (boost_find_component)
  /mnt/build/x-wrt/staging_dir/host/share/cmake-3.31/Modules/FindBoost.cmake:610 (find_package)
  CMakeLists.txt:61 (find_package)

-- Configuring incomplete, errors occurred!
Makefile:88: recipe for target '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot/.configured_92f547ad53268f58943b700581a7a951' failed
make[2]: *** [/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot/.configured_92f547ad53268f58943b700581a7a951] Error 1
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/openwrt-feeds/packages/admin/schroot'
time: package/feeds/packages/schroot/compile#1.40#0.45#1.80
    ERROR: package/feeds/packages/schroot failed to build.
package/Makefile:181: recipe for target 'package/feeds/packages/schroot/compile' failed
make[1]: *** [package/feeds/packages/schroot/compile] Error 1
make[1]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt'
/mnt/Data/Sources/openwrt/x-wrt/include/toplevel.mk:231: recipe for target 'package/feeds/packages/schroot/compile' failed
make: *** [package/feeds/packages/schroot/compile] Error 2

```

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2025-09-03:
clean and rebuild log:
```
find /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.33/ipkg-install/usr/sbin/ -type l -exec cp -fpR -a {} /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.33/.pkgdir/lvm2/sbin/ \;
touch /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.33/.pkgdir/lvm2.installed
mkdir -p /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /mnt/Data/Sources/openwrt/x-wrt/tmp/.root-copy.flock -c 'cp -fpR /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.33/.pkgdir/lvm2/. /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl/root-x86/'
touch /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl/root-x86/stamp/.lvm2_installed
echo "lvm2" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl/pkginfo/lvm2.normal.install
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/openwrt-feeds/packages/utils/lvm2'
time: package/feeds/packages/lvm2/normal/compile#0.96#0.55#1.46
make[2]: Entering directory '/mnt/Data/Sources/openwrt/openwrt-feeds/packages/admin/schroot'
rm -f /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot/.configured_*
rm -f /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl/stamp/.schroot_installed
mkdir -p /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot/build
(cd /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot/build; CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot=reschroot -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot=reschroot -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " LDFLAGS="-L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/usr/lib -L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/lib -fuse-ld=bfd -znow -zrelro   " cmake --no-warn-unused-cli -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=x86_64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/usr/lib -L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/lib -fuse-ld=bfd -znow -zrelro  " -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/usr/lib -L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/lib -fuse-ld=bfd -znow -zrelro   -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/usr/lib -L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/lib -fuse-ld=bfd -znow -zrelro   -Wl,-Bsymbolic-functions" -DCMAKE_AR="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl/usr;/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-x86_64_gcc-14.3.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl -DCMAKE_PREFIX_PATH=/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-x86_64_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE -DCMAKE_GENERATOR="Ninja" -Dbtrfs-snapshot=ON -Dloopback=ON -Dlvm-snapshot=ON -Duuid=ON -Dblock-device=ON -Dunion=ON -Ddhcroot=OFF -Ddchroot-dsa=OFF -Ddebug=OFF -Ddoxygen=OFF -Dpam=OFF -Dtest=OFF -Duuid=OFF -Dzfs-snapshot=OFF /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-x86_64_musl/reschroot-1.6.13/reschroot )
Not searching for unused variables given on the command line.
CMake Deprecation Warning at CMakeLists.txt:16 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

[... cut ...]
time: package/feeds/packages/schroot/compile#0.11#0.07#0.17
    ERROR: package/feeds/packages/schroot failed to build.
package/Makefile:181: recipe for target 'package/feeds/packages/schroot/compile' failed
make[1]: *** [package/feeds/packages/schroot/compile] Error 1
make[1]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt'
/mnt/Data/Sources/openwrt/x-wrt/include/toplevel.mk:231: recipe for target 'package/feeds/packages/schroot/compile' failed
make: *** [package/feeds/packages/schroot/compile] Error 2

```

BKPepe, 2025-09-24:
Oh, I created similar issue for schroot https://github.com/openwrt/packages/issues/27504, there is also patch to fix it