openwrt/packages #27437: pbr: custom proto tunnels can't be used

dmitrmax, 2025-09-13:
### Package Name

pbr

### Maintainer

Stan Grishin

### OpenWrt Version

24.10.2

### OpenWrt Target/Subtarget

ramips/mt7621

### Steps to Reproduce

1. Add interface with custom tunneling proto, e.g. amneziawg
2. Execut `ubus -S call luci.pbr getInterfaces` command.
3. Observe that there is no added amneziawg interface in the list of supported interface for PBR

### Actual Behaviour

/etc/init.d/pbr contains some code to determine the supported interfaces types. This method is hardcoded and doesn't allow to add custom protocol types.

It would be cool if script contains some sort of hook which can be used to install additional criteria by other packages providing custom tunneling protos.

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

stangri, 2025-09-14:
README clearly states supported protocols.

> It would be cool if script contains some sort of hook which can be used to install additional criteria by other packages providing custom tunneling protos.

It would be, do you want to code that?

Alternatively, maintainers/submitters of the new protocols can add a one-liner to the init script of pbr to enable support and submit PR to the upstream repo. Same result with much less effort.

Ser9ei, 2025-09-16:
In addition to the comment above.
This is just a small change in your local repository or directly on the router: replace the relevant lines in the /etc/init.d/pbr file with the snippet below.
There’s no need to submit this to the official repository, as the protocol is not supported.

https://github.com/openwrt/packages/blob/453b3604ce0c3e964e47d1bfbda7eabfb383923f/net/pbr/files/etc/init.d/pbr#L291
https://github.com/openwrt/packages/blob/453b3604ce0c3e964e47d1bfbda7eabfb383923f/net/pbr/files/etc/init.d/pbr#L292

> is_wg() { local p lp; network_get_protocol p "$1"; uci_get_listen_port lp "$1"; [ -z "$lp" ] && [ "${p:0:9}" = "wireguard" -o "${p:0:9}" = "amneziawg" ]; }

> is_wg_server() { local p lp; network_get_protocol p "$1"; uci_get_listen_port lp "$1"; [ -n "$lp" ] && [ "${p:0:9}" = "wireguard" -o "${p:0:9}" = "amneziawg" ]; }

dmitrmax, 2025-09-16:
@Ser9ei I did it even before I created the issue. But I believe I get problems on the next upgrade. That is why I'd rather see some generic way upstream.

@stangri Ok, I'll do it. I see this as a directory like /etc/pbr.d/ where other protocol packages can place their script to check criteria if the interface is a good tunnel for pbr. Are you OK with this way and the directory path/name?

stangri, 2025-09-17:
@dmitrmax `/etc/pbr.d/` is already in use for other purposes. I'd highly recommend you send PRs to the upstream repo, not here.