openwrt/packages #27486: ddns-dnspod_cn_v3: can't use wildcard(*) in domain

tjc0726, 2025-09-20:
### Package Name

ddns

### Maintainer

(@rbqvq) (@feckert)

### OpenWrt Version

24.10.2

### OpenWrt Target/Subtarget

24.10.2

### Steps to Reproduce

open ddns in Luci app
select dnspod_cn_v3 as ddns provider
fill domain in format like *.xxx.yyy
since the the domain should be url format, so i tried %2A.xxx.yyy

### Actual Behaviour

with *.xxx.yyy
```
 203111       : #> /usr/bin/curl -RsS -o /var/run/ddns/a.dat --stderr /var/run/ddns/a.err --interface br-lan --noproxy '*' --request POST https://dnspod.tencentcloudapi.com --data '{"Domain":"*.xxx.yyy","Offset":0,"Limit":1,"RecordType":"A","Subdomain":"*.xxx.yyy"}' --header 'HOST: dnspod.tencentcloudapi.com' --header 'Content-Type: application/json' --header 'X-TC-Action: DescribeRecordList' --header 'X-TC-Version: 2021-03-23' --header 'X-TC-Timestamp: 1758371469' --header 'Authorization: TC3-HMAC-SHA256 Credential=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx/2025-09-20/dnspod/tc3_request, SignedHeaders=content-type;host;x-tc-action, Signature=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx' 
 203112 ERROR : cURL Response Error: 'InvalidParameter.DomainInvalid'
 203112       : {"Response":{"Error":{"Code":"InvalidParameter.DomainInvalid","Message":"u57dfu540du4e0du6b63u786euff0cu8bf7u8f93u5165u4e3bu57dfu540duff0cu5982 dnspod.cnu3002"},"RequestId":"9b778899-22ce-4fd4-8add-f5b9435e0d5c"}}
```
with %2A.xxx.yyy
```
 204145       : ************ ************** ************** **************
 204145  note : PID '21546' started at 2025-09-20 20:41
 204145       : ddns version  : 2.8.2-49
 204145       : uci configuration:
ddns.a.bind_network='lan'
ddns.a.check_unit='minutes'
ddns.a.domain='%2A.xxx.yyy'
ddns.a.enabled='1'
ddns.a.force_unit='minutes'
ddns.a.interface='br-lan'
ddns.a.ip_source='web'
ddns.a.ip_url='http://checkip.dyndns.com'
ddns.a.lookup_host='aa.xxx.yyy'
ddns.a.password='***PW***'
ddns.a.retry_unit='seconds'
ddns.a.service_name='dnspod.cn-v3'
ddns.a.use_ipv6='0'
ddns.a.use_syslog='2'
ddns.a.username='xxxxxxxxxxxxxx'
ddns.a=service
 204145       : verbose mode  : 0 - run normal, NO console output
 204145  CRIT : sanitize on domain found characters outside allowed subset - TERMINATE
```

**I believe the parameter should be sth like:
"Domain":"xxx.yyy","Subdomain":"*(%2A if need url format)"
instead of
"Domain":"*.xxx.yyy","Subdomain":"*.xxx.yyy"**

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

rbqvq, 2025-09-20:
I'm not maintainer :(

tjc0726, 2025-09-20:
> I'm not maintainer :(

sry for that, already corrected it