openwrt/packages #27516: kea: failure to build related to Boost libraries

pprindeville, 2025-09-24:
### Package Name

kea

### Maintainer

@rosysong

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

x86/64

### Steps to Reproduce

1. Try to build `kea`

### Actual Behaviour

The build fails the `autoconf` stage with:

```
...
2025-09-22T22:13:57.4813050Z autoreconf: configure.ac: not using Intltool
2025-09-22T22:13:57.4815182Z autoreconf: configure.ac: not using Gtkdoc
2025-09-22T22:13:57.4817295Z autoreconf: running: /builder/staging_dir/host/bin/aclocal -I /builder/staging_dir/target-x86_64_musl/host/share/aclocal -I /builder/staging_dir/hostpkg/share/aclocal -I /builder/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . --force -I m4macros ${ACLOCAL_FLAGS}
2025-09-22T22:14:04.9028459Z autoreconf: running: /builder/staging_dir/host/bin/autoconf --include=/builder/staging_dir/target-x86_64_musl/host/share/aclocal --include=/builder/staging_dir/hostpkg/share/aclocal --include=/builder/staging_dir/target-x86_64_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/builder/staging_dir/host/share/aclocal --force
2025-09-22T22:14:05.6059044Z autoreconf: running: /builder/staging_dir/host/bin/autoheader --include=/builder/staging_dir/target-x86_64_musl/host/share/aclocal --include=/builder/staging_dir/hostpkg/share/aclocal --include=/builder/staging_dir/target-x86_64_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/builder/staging_dir/host/share/aclocal --force
2025-09-22T22:14:06.1590358Z autoreconf: running: /builder/staging_dir/host/bin/automake --add-missing --copy --force-missing
2025-09-22T22:14:12.2280467Z autoreconf: Leaving directory '.'
2025-09-22T22:14:12.2319232Z (cd /builder/build_dir/target-x86_64_musl/kea-2.6.4/./; if [ -x ./configure ]; then find /builder/build_dir/target-x86_64_musl/kea-2.6.4/ -name config.guess | xargs -r chmod u+w; find /builder/build_dir/target-x86_64_musl/kea-2.6.4/ -name config.guess | xargs -r -n1 cp --remove-destination /builder/scripts/config.guess; find /builder/build_dir/target-x86_64_musl/kea-2.6.4/ -name config.sub | xargs -r chmod u+w; find /builder/build_dir/target-x86_64_musl/kea-2.6.4/ -name config.sub | xargs -r -n1 cp --remove-destination /builder/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/builder/build_dir/target-x86_64_musl/kea-2.6.4=kea-2.6.4 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="x86_64-openwrt-linux-musl-ld.bfd" NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/builder/build_dir/target-x86_64_musl/kea-2.6.4=kea-2.6.4 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/builder/build_dir/target-x86_64_musl/kea-2.6.4=kea-2.6.4 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -DPIC -fpic -std=c++17 " CPPFLAGS="-I/builder/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/usr/include -I/builder/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/include -I/builder/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/include/fortify " LDFLAGS="-L/builder/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/usr/lib -L/builder/staging_dir/toolchain-x86_64_gcc-14.3.0_musl/lib -fuse-ld=bfd -Wl,--gc-sections -znow -zrelro -Wl,--as-needed "  cross_compiling="yes"  ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --with-boost-include="/builder/staging_dir/target-x86_64_musl/usr" --with-log4cplus="/builder/staging_dir/target-x86_64_musl/usr" --with-openssl="/builder/staging_dir/target-x86_64_musl/usr" --without-pic --enable-perfdhcp --enable-shell ; fi; )
2025-09-22T22:14:12.3900843Z configure: WARNING: unrecognized options: --disable-nls
2025-09-22T22:14:12.4258237Z configure: loading site script /builder/include/site/x86_64
2025-09-22T22:14:12.4384779Z checking for a BSD-compatible install... /builder/staging_dir/host/bin/ginstall -c
2025-09-22T22:14:12.4406383Z checking whether sleep supports fractional seconds... yes
2025-09-22T22:14:12.5205689Z checking filesystem timestamp resolution... 0.01
2025-09-22T22:14:12.5240971Z checking whether build environment is sane... yes
2025-09-22T22:14:12.5313825Z checking for x86_64-openwrt-linux-strip... x86_64-openwrt-linux-musl-strip
2025-09-22T22:14:12.5337445Z checking for a race-free mkdir -p... /bin/mkdir -p
2025-09-22T22:14:12.5341804Z checking for gawk... gawk
2025-09-22T22:14:12.5428570Z checking whether make sets $(MAKE)... yes
2025-09-22T22:14:12.5512769Z checking whether make supports nested variables... yes
2025-09-22T22:14:12.5598917Z checking xargs -n works... yes
2025-09-22T22:14:12.5650273Z checking whether UID '1000' is supported by ustar format... yes
2025-09-22T22:14:12.5652000Z checking whether GID '1000' is supported by ustar format... yes
2025-09-22T22:14:12.5844603Z checking how to create a ustar tar archive... gnutar
2025-09-22T22:14:12.5983527Z checking build system type... x86_64-pc-linux-gnu
2025-09-22T22:14:12.6091668Z checking host system type... x86_64-openwrt-linux-gnu
2025-09-22T22:14:12.6920237Z checking whether the C++ compiler works... yes
2025-09-22T22:14:12.6920689Z checking for C++ compiler default output file name... a.out
2025-09-22T22:14:12.7350051Z checking for suffix of executables... 
2025-09-22T22:14:12.7370361Z checking whether we are cross compiling... yes
2025-09-22T22:14:12.7619565Z checking for suffix of object files... o
2025-09-22T22:14:12.7851891Z checking whether the compiler supports GNU C++... yes
2025-09-22T22:14:12.8125102Z checking whether x86_64-openwrt-linux-musl-g++ accepts -g... yes
2025-09-22T22:14:12.8787594Z checking for x86_64-openwrt-linux-musl-g++ option to enable C++11 features... none needed
2025-09-22T22:14:12.8864721Z checking whether make supports the include directive... yes (GNU style)
2025-09-22T22:14:12.8867844Z checking dependency style of x86_64-openwrt-linux-musl-g++... none
2025-09-22T22:14:12.9235674Z checking how to run the C++ preprocessor... x86_64-openwrt-linux-musl-g++ -E
2025-09-22T22:14:12.9606133Z checking whether this is a tarball or git source or package preparation... "tarball"
2025-09-22T22:14:12.9632662Z checking whether this is a development or stable version... 0
2025-09-22T22:14:12.9710520Z checking for pkg-config... /builder/staging_dir/host/bin/pkg-config
2025-09-22T22:14:12.9770982Z checking how to print strings... printf
2025-09-22T22:14:12.9773716Z checking for x86_64-openwrt-linux-gcc... x86_64-openwrt-linux-musl-gcc
2025-09-22T22:14:13.0440858Z checking whether the compiler supports GNU C... yes
2025-09-22T22:14:13.1366542Z checking whether x86_64-openwrt-linux-musl-gcc accepts -g... yes
2025-09-22T22:14:13.1838202Z checking for x86_64-openwrt-linux-musl-gcc option to enable C23 features... -std=gnu23
[... 134 lines cut ...]
2025-09-22T22:14:25.7865393Z checking for boost/system/error_code.hpp... yes
2025-09-22T22:14:25.8275661Z checking for boost/asio/coroutine.hpp... yes
2025-09-22T22:14:26.1058567Z checking for boost/integer/common_factor.hpp... yes
2025-09-22T22:14:26.9556147Z checking for boost/asio/io_context.hpp... yes
2025-09-22T22:14:26.9844038Z checking whether __clang__ is declared... (cached) no
2025-09-22T22:14:27.0225155Z checking whether Boost tries to use threads... yes
2025-09-22T22:14:27.0926260Z checking Boost offset_ptr compiles... yes
2025-09-22T22:14:27.5907272Z checking Boost numeric_cast compiles with -Werror... yes
2025-09-22T22:14:27.6308715Z checking BOOST_STATIC_ASSERT compiles... yes
[... cut ...]
2025-09-22T22:14:28.7872236Z make[1]: Leaving directory '/builder'
2025-09-22T22:14:28.7878594Z make: *** [/builder/include/toplevel.mk:226: package/kea/compile] Error 2
```

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

pprindeville, 2025-09-24:
See [this build](https://productionresultssa18.blob.core.windows.net/actions-results/af7c4ee5-9d17-4648-85c4-d788bd82dd02/workflow-job-run-3a897824-0b14-5e11-8a35-e92a40222788/logs/job/job-logs.txt?rsct=text%2Fplain&se=2025-09-24T16%3A10%3A08Z&sig=dOrDyW6E7XCXsCym32GuJr57j1iXCdgh4zu1mB8C%2FXo%3D&ske=2025-09-25T02%3A27%3A52Z&skoid=ca7593d4-ee42-46cd-af88-8b886a2f84eb&sks=b&skt=2025-09-24T14%3A27%3A52Z&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skv=2025-11-05&sp=r&spr=https&sr=b&st=2025-09-24T16%3A00%3A03Z&sv=2025-11-05).

ClaymorePT, 2025-09-25:
Hey! Sorry for a later reply... work and stuff..

So, Boost.System is now a Header Only library.
I can have b2 build lboost_system for retro compatibility, but it won't do anything when linked.

Let me know if you guys need me to do this.

pprindeville, 2025-09-29:
What about a patch to the `configure` script so it no longer looks for it?

neheb, 2025-10-19:
Anyway, kea builds now.