openwrt/packages #27523: kea: update to 3.0.1 or 3.1.2

pprindeville, 2025-09-25:
### Package Name

kea

### Maintainer

@rosysong

### OpenWrt Version

SNAPSHOT

### OpenWrt Target/Subtarget

x86/64

### Steps to Reproduce

N/A

### Actual Behaviour

Need an update to 3.0.1 LTS or 3.1.2 LATEST.

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

pprindeville, 2025-09-26:
Note that `Kea` now builds with `meson`.

dave14305, 2025-11-06:
Can this be closed after https://github.com/openwrt/packages/commit/728e9e1073bf54b5e5c987b18d2bdd59be1a4253 was committed for kea 3.0.1?