openwrt/packages #27540: xray-core: Please update to 25.9.11

ofry, 2025-09-27:
### Package Name

xray-core

### Maintainer

Tianling Shen <cnsztl@immortalwrt.org>

### OpenWrt Version

24.10.3

### OpenWrt Target/Subtarget

mediatek/filogic

### Steps to Reproduce

1. Just see `opkg` output.

### Actual Behaviour

Current `xray-core` package version is `25.1.30-r1`

Latest version of `xray-core` is `25.9.11`

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ofry, 2025-09-27:
This is related to https://github.com/XTLS/Xray-core/issues/5173

I can't make the bugreport in Xray-core repo, because it should be reproduced on the latest version of package.

1715173329, 2025-09-28:
You can use snapshot builds instead.