openwrt/packages #27544: golang: update at least to 1.25.1

ofry, 2025-09-28:
### Package Name

golang

### Maintainer

Jeffery To

### OpenWrt Version

24.10.3

### OpenWrt Target/Subtarget

mediatek/filogic

### Steps to Reproduce

1. Just see `opkg` output.

### Actual Behaviour

Current `golang` package version is 1.23.12-r2

It's related to https://github.com/openwrt/packages/issues/27540
Because `Xray-core` v25.9.11 (latest version) requires at least Go 1.25.1

It has some DNS-related improvements compared to `25.1.30-r1`.
And "combo" `xray-core`+`v2raya` is highly unstable on my router.

But `xray-core` devs closed my bug report, because it must be reproduced on latest stable version.

Related to https://github.com/XTLS/Xray-core/issues/5173 and https://github.com/openwrt/packages/issues/27540

### Confirmation Checklist

- [x] The package is maintained in this repository.
- [x] I understand that issues related to [the base OpenWrt repository](https://github.com/openwrt/openwrt/issues) or [LuCI repository](https://github.com/openwrt/luci/issues) will be closed.
- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

1715173329, 2025-09-28:
They also provide prebuilt binaries, you can manually replace it.
Anyway, it's impossible to do such major upgrade in stable branch, at least now and in near future.

closed 2025-09-28 by 1715173329

ofry, 2025-10-20:
@1715173329 Golang 1.25 exists on `SNAPSHOT`. Could it be backported to stable eventually?